microsoft/AirSim #2118: Changing communication frequency between airsim and python api

rmdwjdrhkrkatk, 2019-08-05:
Hi, I'm working on controlling quadrotor using customized python api which commands signal to individual motor.
Since motor needs high frequency of command, I am trying to command motor signal in 250~1000Hz.
However, when I command rotor speed in the duration of 0.005sec(200hz), simulation time elapses about 0.02sec(50hz).
I'm pretty confident that the duration function of customized api is working fine.

I studied the clock system in Airsim, and found out that maximum communication frequency between Airsim and python api is set in 50hz(0.02Hz).
I concluded that if it is possible to syncronize communication frequency with motor signal frequency, (in other words, enhancing communication frequency in the level of 250Hz) I could keep sending signal to quadrotor with no delay.

Is there anyone who changed default settings of source code, or give some solution of the problem?

msb336, 2019-08-07:
You can slow down the simulation clock to achieve higher frequency low level control by changing the clock speed in your settings file:
https://github.com/microsoft/AirSim/blob/master/docs/settings.md#available-settings-and-their-defaults

closed 2020-03-12 by ironclownfish

josepmaib, 2020-05-13:
I am also trying to control each speed using moveByMotorPWMs python api but I found that the communication frequency is not enough. I tried @msb336 solution, slowing down the simulator clock and it works correctly but I need a different solution to have the clock speed at the original speed. 

Somebody has modified communication frequency in the source code?

Really appreciate the help