microsoft/AirSim #2119: AirSim build error with Visual Studio 2017 15.8.9

RinaKomatsu2019, 2019-08-06:
I'm using Windows 10 & Visual Studio 2017 (version:15.8.9). When I tried building AirSim following the instruction, I faced the error like that (Sorry Japanese!). Is there anyone who knows how to solve?

**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.8.9
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim>build.cmd
Found cmake version: 3.10.2
Starting cmake to build rpclib...
-- The C compiler identification is MSVC 19.15.26732.1
-- The CXX compiler identification is MSVC 19.15.26732.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x64/cl.exe -- broken
CMake Error at C:/Program Files/CMake/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x64/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/AirSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeTmp

    Run Build Command:"C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/MSBuild/15.0/Bin/MSBuild.exe" "cmTC_398c6.vcxproj" "/p:Configuration=Debug" "/p:VisualStudioVersion=15.0"
    .NET Framework 向け Microsoft (R) Build Engine バージョン 15.8.169+g1ccb72aefa
    Copyright (C) Microsoft Corporation.All rights reserved.

    2019/08/06 12:50:34 にビルドを開始しました。
    ノード 1 上のプロジェクト "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj" (既定のターゲット)。
    PrepareForBuild:
      ディレクトリ "cmTC_398c6.dir\Debug\" を作成しています。
      ディレクトリ "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\Debug\" を作成しています。
      ディレクトリ "cmTC_398c6.dir\Debug\cmTC_398c6.tlog\" を作成しています。
    InitializeBuildStatus:
      "AlwaysCreate" が指定されたため "cmTC_398c6.dir\Debug\cmTC_398c6.tlog\unsuccessfulbuild" を作成しています。
    ClCompile:
      C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX86\x64\CL.exe /c /Zi /W3 /WX- /diagnostics:classic /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_398c6.dir\Debug\\" /Fd"cmTC_398c6.dir\Debug\vc141.pdb" /Gd /TC /FC /errorReport:queue  C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v9.1\include "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\testCCompiler.c"
      Microsoft(R) C/C++ Optimizing Compiler Version 19.15.26732.1 for x64
      Copyright (C) Microsoft Corporation.  All rights reserved.

      Program
      cl /c /Zi /W3 /WX- /diagnostics:classic /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_398c6.dir\Debug\\" /Fd"cmTC_398c6.dir\Debug\vc141.pdb" /Gd /TC /FC /errorReport:queue  C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v9.1\include "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\testCCompiler.c"
    c1 : fatal error C1083: ソース ファイルを開けません。'C:\Program':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]

      NVIDIA
    c1 : fatal error C1083: ソース ファイルを開けません。'Files\NVIDIA':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]
      GPU
    c1 : fatal error C1083: ソース ファイルを開けません。'GPU':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]
      Computing
    c1 : fatal error C1083: ソース ファイルを開けません。'Computing':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]
      include
    c1 : fatal error C1083: ソース ファイルを開けません。'Toolkit\CUDA\v9.1\include':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]
      testCCompiler.c
      コードを生成中...
    プロジェクト "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj" (既定のターゲット) のビルドが終了しました -- 失敗。

    ビルドに失敗しました。

    "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj" (既定のターゲット) (1) ->
    (ClCompile ターゲット) ->
      c1 : fatal error C1083: ソース ファイルを開けません。'C:\Program':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]
      c1 : fatal error C1083: ソース ファイルを開けません。'Files\NVIDIA':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]
      c1 : fatal error C1083: ソース ファイルを開けません。'GPU':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]
      c1 : fatal error C1083: ソース ファイルを開けません。'Computing':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]
      c1 : fatal error C1083: ソース ファイルを開けません。'Toolkit\CUDA\v9.1\include':No such file or directory [C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeTmp\cmTC_398c6.vcxproj]

        0 個の警告
        5 エラー

    経過時間 00:00:00.67

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/AirSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeOutput.log".
See also "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/AirSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeError.log".
.NET Framework 向け Microsoft (R) Build Engine バージョン 15.8.169+g1ccb72aefa
Copyright (C) Microsoft Corporation.All rights reserved.

MSBUILD : error MSB1009: プロジェクト ファイルが存在しません。
スイッチ: ALL_BUILD.vcxproj
.NET Framework 向け Microsoft (R) Build Engine バージョン 15.8.169+g1ccb72aefa
Copyright (C) Microsoft Corporation.All rights reserved.

MSBUILD : error MSB1009: プロジェクト ファイルが存在しません。
スイッチ: ALL_BUILD.vcxproj

#### Build failed - see messages above.

aesopcode, 2019-10-23:
I meet the same error about "msbuild", have you solved it yet?

aesopcode, 2019-10-23:
you can add "C:\Windows\Microsoft.NET\Framework64\v4.0.30319" or "C:\Windows\Microsoft.NET\Framework\v4.0.30319" (according to x64 or x32) to your system environment path.

Rinax2019, 2019-10-23:
Dear aesopcode

 Thank you for replying! 
 Unfortunately, the problem has not be solved yet. Seeing the error code, it seems there is the problem that the project file "ALL_BUILD.vcxproj" can not be found.
 I'll reply again after add each path to environment path "PATH" and try to build. Thank you!

Rinax2019, 2019-10-23:
Dear aesopcode

 Follwed your advice, I tried buliding again with VS2019. The error with "ALL_BUILD.vcxproj" is solved! However, I met another error  like below image. (At first, I met the error about AirSim\AirLib\deps\eigen3\Eigen\src\Core\arch\CUDA\Half.h, I could solve by save as UTF-8 (inclide BOM) ) Is it seems problem with VS2019?

[image]

aesopcode, 2019-10-23:
Well, I also had a problem, but it is a different error. I will put a picture which records the detailed information at the following comment(There are some Chinese words that you can ignore them). I searched some anwsers pointing out that it's about VS C++ Build tools, so I am downloading them. You could try to look for some advice on stackoverflow.
[image]

Rinax2019, 2019-10-23:
Dear aesopcode

  I tried building AirSim after re-install VS2017 with VS C++ Build tools and Windows SDK 8. When I typed "build.cmd" at x64 Native Tools Command Prompt for VS 2017, AirSim suceeded to build! Thank you for your advice! I think my problem related with the enviroment path & Visual Studio version.
 Seeing your picture, I think it outputs error when try to build using "Command Prompt".  Find "x64 Native Tools Command Prompt for VS 2017" and try to build!
 Thank you for all.

aesopcode, 2019-10-23:
Thank you for your advice! I have built AirSim successfully by using x64 Native Tools Command Prompt.
Good luck!

closed 2019-10-23 by Rinax2019