microsoft/AirSim #2121: Mac build - CMAKE_C_COMPILER not set, after EnableLanguage

lloydrayner, 2019-08-06:
Hi, 

Hope you can help. 

I have managed to get ./setup.sh to run but when trying to run ./build.sh I get the below error. I am running on a Mac with Mojave.

Any help would be appreciated

```
putting build in build_debug folder, to clean, just delete the directory...
CMake Error at /Users/lrayner/PycharmProjects/T1.V1/UnrealEngine/AirSim/cmake_build/Modules/CMakeDetermineCCompiler.cmake:48 (message):
  Could not find compiler set in environment variable CC:

  /usr/local/opt/llvm-5.0/bin/clang-5.0.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/Users/lrayner/PycharmProjects/T1.V1/UnrealEngine/AirSim/build_debug/CMakeFiles/CMakeOutput.log".
~/PycharmProjects/T1.V1/UnrealEngine/AirSim ~/PycharmProjects/T1.V1/UnrealEngine/AirSim

```

lloydrayner, 2019-08-10:
Figured it out.

closed 2019-08-10 by lloydrayner

atul337, 2020-04-03:
Can you help me with how to fix this?

sharath29, 2023-08-14:
> Figured it out.

This person is likes to see other developer suffer.