microsoft/AirSim #2123: Trained file issue

afirahzahid, 2019-08-07:
Hi,
     I am working on drones in air-sim using reinforcement learning. Whenever I run my code, training begins but I don't know where its trained model is placed in air-sim or in my PC and don't know how to get back that trained model in air-sim for further training(don't know about load and save of trained file). So kindly help me in this issue. I am badly stuck here.

msb336, 2019-08-07:
Airsim doesn't explicitly do training, we only provide an interface for you to train your models. Saving and retrieving models is dependent on what machine learning package you are using.

sillycornvalley, 2019-08-12:
But you can train the model using DQNcar.py/DQNDdrone.py, right?

msb336, 2019-08-12:
Correct. I don't think either of the DQN demo scripts save their models anywhere, but you can do so by following the steps illustrated:
```
from cntk import persist
persist.save_model({your_model}, "{your_file}.dnn")
```

closed 2019-08-22 by msb336