microsoft/AirSim #2127: Can't open Blocks.sh in SITL in ubuntu 16.04. Just  can use UE4 editor in sitl?

machenxiang, 2019-08-08:
(base) a210@210:~/AirSim_environment/Blocks$ ./Blocks.sh 
Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogInit: Display: RandInit(808844551) SRandInit(808844551).
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.021568
LogPluginManager: Mounting plugin AirSim
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin AppleARKit
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 
LogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
LogInit: Build: ++UE4+Release-4.18-CL-0
LogInit: Engine Version: 4.18.3-0+++UE4+Release-4.18
LogInit: Compatible Engine Version: 4.18.0-0+++UE4+Release-4.18
LogInit: Net CL: 0
LogDevObjectVersion: Number of dev versions registered: 16
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0
LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 2
LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 20
LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 30
LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 0
LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0
LogDevObjectVersion:   Dev-Online (39C831C9-5AE6-47DC-9A44-9C173E1C8E7C): 0
LogDevObjectVersion:   Dev-Physics (78F01B33-EBEA-4F98-B9B4-84EACCB95AA2): 0
LogDevObjectVersion:   Dev-Platform (6631380F-2D4D-43E0-8009-CF276956A95A): 0
[... 510 lines cut ...]
[2019.08.08-11.06.11:803][  0]LogTemp: Warning: Warning, WeatherAPI got invalid weather actor class!
[2019.08.08-11.06.11:803][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Weather/UI/MenuActor requested by async loading code. NameToLoad: /AirSim/Weather/UI/MenuActor
[2019.08.08-11.06.11:803][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.08.08-11.06.11:803][  0]LogTemp: Warning: Warning, WeatherAPI got invalid menu actor class!
[2019.08.08-11.06.11:803][  0]LogTemp: PX4
PX4
[2019.08.08-11.06.11:803][  0]LogUMG: Display: Widget Class BP_SimHUDWidget_C - Loaded Fast Template.
[2019.08.08-11.06.11:804][  0]LogLoad: Took 0.161615 seconds to LoadMap(/Game/FlyingCPP/Maps/FlyingExampleMap)
[2019.08.08-11.06.11:806][  0]LogSlate: Took 0.001815 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
[2019.08.08-11.06.11:948][  0]LogSlate: Took 0.001875 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoRegular.ufont' (155K)
[2019.08.08-11.06.11:952][  0]LogLoad: (Engine Initialization) Total time: 2.73 seconds
[2019.08.08-11.06.11:952][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2019.08.08-11.06.12:007][  0]LogRenderer: Reallocating scene render targets to support 1920x1080 Format 10 NumSamples 1 (Frame:1).
[2019.08.08-11.06.12:082][  0]LogInit: Adding external input plugin.
[2019.08.08-11.06.12:084][  0]LogContentStreaming: Texture pool size now 1000 MB
[2019.08.08-11.06.12:173][  1]LogLinux: Setting swap interval to 'Immediate'
terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 111

Signal 6 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=196655 
Malloc Size=42799 LargeMemoryPoolOffset=239471 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)

qratosone, 2019-08-21:
We are facing the same problems when we deploy the UE4 project with the latest version of AirSim. Now we can only run airsim with PX4 SITL inside the UE4 Editor

machenxiang, 2019-08-23:
> We are facing the same problems when we deploy the UE4 project with the latest version of AirSim. Now we can only run airsim with PX4 SITL inside the UE4 Editor

Thanks for your reply, I'm also using UE4 Editor. But the UE4 editor only has Blocks, how can I use other secens？

qratosone, 2019-09-01:
> > We are facing the same problems when we deploy the UE4 project with the latest version of AirSim. Now we can only run airsim with PX4 SITL inside the UE4 Editor
> 
> Thanks for your reply, I'm also using UE4 Editor. But the UE4 editor only has Blocks, how can I use other secens？

just do as the document says

https://microsoft.github.io/AirSim/docs/unreal_custenv/

lovettchris, 2019-11-17:
Can you try my latest pull request, it fixes a lot of issues with PX4 SITL:
https://github.com/microsoft/AirSim/pull/2310

qratosone, 2019-11-20:
> Can you try my latest pull request, it fixes a lot of issues with PX4 SITL:
> #2310

OK，I'll try this.

qratosone, 2019-12-21:
> Can you try my latest pull request, it fixes a lot of issues with PX4 SITL:
> #2310

Got another error code

[2019.12.21-00.58.17:139][  0]LogTemp: Disconnecting mavlink vehicle
Disconnecting mavlink vehicle
Waiting for mavlink vehicle...
Disconnecting mavlink vehicle
Connected to SITL over UDP.
Connecting to PX4 Control UDP port 14580, local IP 127.0.0.1, remote IP...
[2019.12.21-00.58.17:139][  0]LogTemp: Waiting for mavlink vehicle...
[2019.12.21-00.58.17:139][  0]LogTemp: Disconnecting mavlink vehicle
[2019.12.21-00.58.17:139][  0]LogTemp: Connected to SITL over UDP.
[2019.12.21-00.58.17:139][  0]LogTemp: Connecting to PX4 Control UDP port 14580, local IP 127.0.0.1, remote IP...
[2019.12.21-00.58.17:139][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Weather/WeatherFX/WeatherActor requested by async loading code. NameToLoad: /AirSim/Weather/WeatherFX/WeatherActor
[2019.12.21-00.58.17:139][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.12.21-00.58.17:139][  0]LogTemp: Warning: Warning, WeatherAPI got invalid weather actor class!
[2019.12.21-00.58.17:139][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Weather/UI/MenuActor requested by async loading code. NameToLoad: /AirSim/Weather/UI/MenuActor
[2019.12.21-00.58.17:139][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.12.21-00.58.17:139][  0]LogTemp: Warning: Warning, WeatherAPI got invalid menu actor class!
[2019.12.21-00.58.17:140][  0]LogTemp: PX4
PX4
[2019.12.21-00.58.17:140][  0]LogUMG: Display: Widget Class BP_SimHUDWidget_C - Loaded Fast Template.
[2019.12.21-00.58.17:141][  0]LogLoad: Took 0.291492 seconds to LoadMap(/Game/FlyingCPP/Maps/FlyingExampleMap)
[2019.12.21-00.58.17:143][  0]LogSlate: Took 0.002241 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
[2019.12.21-00.58.17:149][  0]LogSlate: Took 0.002553 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoRegular.ufont' (155K)
Checking Pak Config[2019.12.21-00.58.17:156][  0]LogLoad: (Engine Initialization) Total time: 4.29 seconds
[2019.12.21-00.58.17:216][  0]LogRenderer: Reallocating scene render targets to support 1920x1080 Format 10 NumSamples 1 (Frame:1).
[2019.12.21-00.58.17:227][  0]LogRHI: Using OpenGL program LRU cache: 0
Ground control connected over UDP.
[2019.12.21-00.58.19:142][  0]LogTemp: Ground control connected over UDP.
terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 9

Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=132912 LargeMemoryPoolOffset=264048 
Failed to find symbol file, expected location:
"/home/qratosone/deploy/LinuxNoEditor/Blocks/Binaries/Linux/Blocks.sym"
Malloc Size=62037 LargeMemoryPoolOffset=326101 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)

saifullah3396, 2019-12-30:
[See this](https://github.com/microsoft/AirSim/pull/2310#issuecomment-569588704)

closed 2021-04-22 by jonyMarino