microsoft/AirSim #2129: Camera FOV becomes unrealistic above 120 degrees

M-J-Murray, 2019-08-08:
Hello

I am attempting to simulate a camera on the default drone with airsim that is as close as possible to a camera I want to use in real-life. However, I'm finding that the FOV functionality in airsim isn't suitable for fisheye lenses. I want to use a 185 degrees FOV on my camera, however, when I configure the simulation to use these settings the results are bizarre, and highly unrealistic. Everything becomes warped and the camera perspective is completely wrong.
I have captured screen shots of what the ue4 shows vs what the simulated camera shows (this is for a custom 3d environment).
[image]
[image]
You can probably guess which one is which. Is there any way the FOV calculations could be re-written so that they are more realistic?
Let me know if you require any more information.

alwynmathew, 2020-01-30:
Can we model a 180 degree fisheye camera in AirSim? @saihv

petergerten, 2020-10-26:
@M-J-Murray I am trying to get a 220 deg fisheye camera to work and get similar results. Could you eventually solve this?