microsoft/AirSim #2131: How to change the publishing rate of airsim_node ?

jzstudent, 2019-08-09:
------------------------------------
when I used the ros package ( airsim_ros_pkgs ) to publish the image or depth image of camera,I found that the rate is only 0.6! How to improve it?

saihv, 2020-01-20:
What image resolution are you using?

Omar-Elmofty, 2020-05-26:
I have the same question, was anyone able to figure out how to change the publishing rate?

GimpelZhang, 2020-05-26:
> I have the same question, was anyone able to figure out how to change the publishing rate?

First please check if the "update_airsim_img_response_every_n_sec" parameter has been tuned. 

If the publishing rate is limited by the computer performance, maybe changing the ClockSpeed setting to a less than one number is a solution. 

I have tested that slower simulation would produce denser image response.

jzstudent, 2020-05-26:
> I have the same question, was anyone able to figure out how to change the publishing rate?

Hi, I've tried to set "ViewMode": "NoDisplay" in .json, it makes the rate improve and I 've already slove it.

Omar-Elmofty, 2020-05-26:
Thanks! The NoDisplay mode helped a lot, also slowing down the simulation helps.

closed 2020-09-04 by jonyMarino