microsoft/AirSim #2134: Aircraft to collide with walls but not each other?

Dan-Moss, 2019-08-13:
Is it possible to allow the aircraft to collide with walls and the ground, but not other vehicles? Basically so that they can overlap? Running in Multi HITL Mode.

msb336, 2019-09-26:
If you have access to the editor for your environment, you can modify the [collision channels](https://www.unrealengine.com/en-US/blog/collision-filtering) in the `FlyingPawn` blueprint so that the drones do not trigger collision events with each other. This change is not accomplishable if you do not have access to the environment editor.

Dan-Moss, 2019-09-27:
Hi, I do have access to the editor. Thank you for the link. From the pawn blueprint how do i get to the collision setting, i cant seem to find it anymore?

msb336, 2019-09-27:
Go to the viewport window, and you will see the drone model. click the model, and on the righthand side there will be a menu category called "collision"

closed 2020-01-20 by saihv