microsoft/AirSim #2136: How to export car/drone state to .csv or .txt using Python API?

FishAres, 2019-08-14:
I want to use the Python API to record the vehicle state iteratively, but I'm having trouble with the "CarState" file format. Is there a straightforward way to write each timestamp's state info to file with the API? Thanks!

ironclownfish, 2019-09-25:
You can use simGetVehiclePose() in a loop in python to repeatedly log the position of the vehicle.
This API returns a Pose object, which has a member (position) that's just the position vector. You could log these into a csv from python.

msb336, 2019-09-26:
@FishAres, [`CarState`](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/types.py#L331-L339) has a relatively straight forward format, so all you would need to do is write a function which logs your state to a file:

```
def logCarState(file_name, car_state):
    with f as open(file_name, 'a'):
        f.write("{},{},{},{},{},{},{} ........{}\n".format(speed, gear,rpm, etc.)
```

closed 2020-01-20 by saihv