microsoft/AirSim #2139: Running AirSim and Unreal on a remote Linux server

ivannson, 2019-08-16:
Hello, 

I was wondering if there are any guides on how to run AirSim and Unreal on a remote Linux server (Ubuntu 18.04)? 

Any guidance would be greatly appreciated.

ivannson, 2019-08-18:
Finally managed to get it working. The problem was with the fact that I didn't realise Microsoft Remote Desktop doesn't allow you to use hardware acceleration for OpenGL 3D rendering, so my dedicated GPU was not used, hence why Unreal wasn't launching. 

I used VirtualGL and TurboVNC to get this working, mainly based on the CARLA Headless tutorial (https://github.com/carla-simulator/carla/blob/master/Docs/carla_headless.md), combined with using Step 2 and Step 3 in Scenario B in this guide: https://summerofhpc.prace-ri.eu/remote-accelerated-graphics-with-virtualgl-and-turbovnc/ .

closed 2019-08-18 by ivannson

janblumenkamp, 2020-03-05:
The Carla headless tutorial was removed from master, but is still available [here](https://github.com/carla-simulator/carla/blob/85b192530dcf8bb6a305c1dc157d32835a542d47/Docs/carla_headless.md). This combined with the second guide worked for me, except I had to set up the xorg.conf according to [these instructions](https://virtualgl.org/Documentation/HeadlessNV) (essentially specify the bus id and leave `--use-display-device=None`).