microsoft/AirSim #2140: Gimbal drift in multirotor

heleidsn, 2019-08-16:
Dear all,

I am doing reinforcement learning work with multirotor. To get a stable image as input of the network, I set the gimbal and stable the Pitch and Roll of the camera. But I find that the gimbal always drifts after hundreds of episodes as shown in the following picture. **Note: one of the arms always appears in the image and I don't set the gimbal for Yaw.** 

I reset the multirotor using **resetClient()** at each end of the episode but it seems doesn't work with the gimbal and leads to a small accumulated drift.

[image]

My setting of Camera:
```
"CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 640,
          "Height": 280,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "AutoExposureBias": -1,
          "AutoExposureMaxBrightness": 0.40,
          "AutoExposureMinBrightness": 0.03,
          "MotionBlurAmount": 0,
          "TargetGamma": 1.0,
          "ProjectionMode": "",
          "OrthoWidth": 5.12
        }
      ],
      "Gimbal": {
        "Stabilization": 1,
        "Pitch": 1,
        "Roll": 1
      }
    }
```
OS：windows 10
AirSim： V1.2.2 master
Python：3.6.5 
Unreal version：4.22.2

Does anyone have the same problem using the gimbal module? Is there any way to solve this problem? 

Thank you very much!

ironclownfish, 2019-10-08:
We can look into this. Can you please share the full settings file so we can make sure our conditions are exactly the same?

heleidsn, 2019-10-09:
Thanks for your reply! This is the full setting file:

    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "ClockSpeed": 1,
    "EngineSound": false,
  
  
    "Recording": {
      "RecordOnMove": true,
      "RecordInterval": 0.05,
      "Cameras": [
        {
          "CameraName": "0",
          "ImageType": 0,
          "PixelsAsFloat": false,
          "Compress": true
        }
      ]
    },
  
    "CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 640,
          "Height": 280,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "AutoExposureBias": -1,
          "AutoExposureMaxBrightness": 0.40,
          "AutoExposureMinBrightness": 0.03,
          "MotionBlurAmount": 0,
          "TargetGamma": 1.0,
          "ProjectionMode": "",
          "OrthoWidth": 5.12
        }
      ],
      "Gimbal": {
        "Stabilization": 1,
        "Pitch": 1,
        "Roll": 1
      }
    },
  
    "Vehicles": {
      "ComputerVision": {
        "VehicleType": "SimpleFlight"
        }
      }
 
I just want to stabilize the pitch and roll axis in setting file. The multirotor is reset using `client.reset()` API when it crashes to obstacles.

DanielGilo, 2020-09-22:
We face the same issue. We only want to stabilize the drone for pitch and roll (we don't want yaw stabilization) for the bottom_center camera (the one that looks down). But when omitting the yaw from the settings.json (only specifying roll and pitch for the gimbal), the camera's yaw moves in an unstable way - doesn't follow the drone's orientation. 

[image]

If the settings are this way, we get stabilization on yaw (not what we want).

[image]

This is supposed to be what we what we want, but as we said the yaw in this case "goes crazy", we just want it to follow the drone's orientation.

Is there any solution?

Zartris, 2020-09-23:
> We face the same issue. We only want to stabilize the drone for pitch and roll (we don't want yaw stabilization) for the bottom_center camera (the one that looks down). But when omitting the yaw [...]

I think this is something separate and you should make a new thread for this.
I have the exact same problem, but i have no solution yet.

msmcconnell, 2023-12-21:
This is still an active issue for me (exactly as other describe above). Has there been any movement on it?

jprodriguezg, 2024-03-28:
I have the same issue reported above. It seems like the gimbal functionality only works while stabilizing all three rotation axes.  I am wondering if someone someone is working on a solution.

honestws, 2025-06-28:
I have the same issue reported above.