microsoft/AirSim #2144: How publish sensor_msgs::LaserScan message ?

machenxiang, 2019-08-20:
I get the lidar date and how to change it into ROS  sensor_msgs::LaserScan message? Thanke in advance.

madratman, 2019-08-22:
https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp#L594-#L636

madratman, 2019-08-22:
although, that's for pointcloud2, not laserscan, as former is 2D, latter is 3D

closed 2019-08-22 by madratman

machenxiang, 2019-08-23:
> although, that's for pointcloud2, not laserscan, as former is 2D, latter is 3D

Thanks for your reply. But I still have some questions.
1. Why is the data get from( lidarData = self.client.getLidarData )different every time when the drone doesn't take off at all?
2.How to get lidar data when in  PX4 SITL？The JSON file is
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode":"Multirotor",
  "Vehicles":{
         "drone_1":{
             	"VehicleType":"PX4Multirotor",
             	"UseSerial":false,
	      "EnableCollisionPassthrogh": false,
	      "EnableCollisions": true,
	      "AllowAPIAlways": true
        },
  
 "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Sensors": {
        "Imu" : {
          "SensorType": 2,
          "Enabled": true
        },
        "LidarCustom": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 1,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "DrawDebugPoints": true
        }
      },
      "Cameras": {
        "front_center_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 27,
              "DepthOfFieldFstop": 2.8,
              "DepthOfFieldFocalDistance": 200.0, 
              "DepthOfFieldFocalRegion": 200.0,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0, "Z": 0.10,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
        "front_left_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }, 
            {
              "PublishToRos": 1,
              "ImageType": 1,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": -0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front_right_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      }
    },
    "SubWindows": [
      {"WindowID": 0, "ImageType": 0, "CameraName": "front_left_custom", "Visible": true},
      {"WindowID": 1, "ImageType": 0, "CameraName": "front_center_custom", "Visible": false},
      {"WindowID": 2, "ImageType": 0, "CameraName": "front_right_custom", "Visible": true}
    ]
}
When i try to get lidar data  use this file it will occur some error.
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "/home/a210/airsim_ros_ws/src/px4_cartg_imu/scripts/pub_lidar_msg.py", line 147, in <module>
    main()
  File "/home/a210/airsim_ros_ws/src/px4_cartg_imu/scripts/pub_lidar_msg.py", line 137, in main
    laser_data=get_lidar.run()
  File "/home/a210/airsim_ros_ws/src/px4_cartg_imu/scripts/pub_lidar_msg.py", line 101, in run
    raw_data=self.getLidarData()
  File "/home/a210/airsim_ros_ws/src/px4_cartg_imu/scripts/pub_lidar_msg.py", line 31, in getLidarData
    lidarData = self.client.getLidarData()
  File "/home/a210/airsim_ros_ws/src/px4_cartg_imu/scripts/airsim/client.py", line 163, in getLidarData
    return LidarData.from_msgpack(self.client.call('getLidarData', lidar_name, vehicle_name))
  File "/home/a210/anaconda3/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/a210/anaconda3/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getLidarData' (called with 2 arg(s)) threw an exception. The exception contained this information: No lidar with name  exist on vehicle.