microsoft/AirSim #2149: Airsim build error

wearesunshines, 2019-08-22:
When building,I met with some question, as is shown following.
What's wrong.

C:\Users\DELL\source\repos\AirSim>build.cmd
Found cmake version: 3.14.19060802-MSVC_2
*****************************************************************************************
Downloading rpclib
*****************************************************************************************
Starting cmake to build rpclib...
CMake Error at CMakeLists.txt:2 (project):
  Generator
    Visual Studio 15 2017 Win64
  could not find any instance of Visual Studio.
-- Configuring incomplete, errors occurred!
See also "C:/Users/DELL/source/repos/AirSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeOutput.log".

CMake Error:
  Generator

    Visual Studio 15 2017 Win64

  could not find any instance of Visual Studio.

#### Build failed - see messages above.

msb336, 2019-08-22:
@wearesunshines
Windows can't find VS 15. Do you have it installed?

wearesunshines, 2019-08-23:
I found the problem, I install VS2019. Thanks!

closed 2019-08-23 by wearesunshines