microsoft/AirSim #2155: Lidar error in sitl,No lidar with name  exist on vehicle.

machenxiang, 2019-08-25:
I use Lidar in AirSim in python. I get the lidardata with 
 def getLidarData():
        lidarData = client.getLidarData()
        return lidarData
But it occurs an error,
Traceback (most recent call last):
  File "pub_lidar_msg.py", line 147, in <module>
    main()
  File "pub_lidar_msg.py", line 137, in main
    laser_data=get_lidar.run()
  File "pub_lidar_msg.py", line 101, in run
    raw_data=self.getLidarData()
  File "pub_lidar_msg.py", line 31, in getLidarData
    lidarData = self.client.getLidarData()
  File "/home/amov/AMOV_WorkSpace/lidar_vfh_ws/src/px4_cartg_imu/scripts/airsim/client.py", line 163, in getLidarData
    return LidarData.from_msgpack(self.client.call('getLidarData', lidar_name, vehicle_name))
  File "/usr/local/lib/python2.7/dist-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/usr/local/lib/python2.7/dist-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getLidarData' (called with 2 arg(s)) threw an exception. The exception contained this information: No lidar with name  exist on vehicle.

And the below is my json file.
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ViewMode": "SpringArmChase",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "PX4Multirotor",
      "UseSerial":false,
   
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true
    },
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Sensors": {
        "Imu" : {
          "SensorType": 2,
          "Enabled": true
        },
        "LidarCustom": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "DrawDebugPoints": true
        }
      },
      "Cameras": {
        "front_center_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 27,
              "DepthOfFieldFstop": 2.8,
              "DepthOfFieldFocalDistance": 200.0, 
              "DepthOfFieldFocalRegion": 200.0,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0, "Z": 0.10,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
        "front_left_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }, 
            {
              "PublishToRos": 1,
              "ImageType": 1,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": -0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front_right_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 2, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    
  },
  "SubWindows": [
    {"WindowID": 0, "ImageType": 1, "CameraName": "front_left_custom", "Visible": true},
    {"WindowID": 1, "ImageType": 2, "CameraName": "front_center_custom", "Visible": false},
    {"WindowID": 2, "ImageType": 0, "CameraName": "front_right_custom", "Visible": true}
  ]
}

weiLiangHare, 2019-08-25:
I have the problem too 
someone hit it?

bmalhigh, 2019-10-01:
Folks -- This problem typically occurs when there is a mismatch of AirSim python client version and  AirSim plugin. Lidar support was added later in the cycle and sometimes folks are still running the old plugin which doesn't have the lidar code. Are you sure you have the matching AirSim plugin running?

chriscreber, 2019-10-13:
I have this issue (for C++ and python) even after reinstalling Unreal and AirSim so are you sure this is just a mismatch problem?

EdwinMeriaux, 2020-06-26:
> Folks -- This problem typically occurs when there is a mismatch of AirSim python client version and AirSim plugin. Lidar support was added later in the cycle and sometimes folks are still running [...]

Hey So I installed Airsim recently (in the last 4 weeks) would it still be my airsim plugin that would be the issue? Would it not (at least for the lidar) be up to date?

jonyMarino, 2020-11-23:
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing an issue
*Attach a screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-11-23 by jonyMarino