microsoft/AirSim #2162: Multirotor follow path algorithm

YumaTheCompanion, 2019-08-28:
It is possible to follow a predetermined path using `moveOnPathAsync`.

I've tried and tested it, it works as expected. However, the yaw angle does not change, the multirotor is facing the same direction from start.

**Is it possible to make the drone face the direction it is moving, (as in car simulation) ?**

**If not, I'm willing to add such a functionality. Is anyone also interested in such a thing?**

YumaTheCompanion, 2019-08-28:
I meant as in; is it possible in core functionality, without implementing it ?

As of now, I've created a wrapper function that uses a combination of `moveOnPathAsync` and `moveByVelocityZAsync` functions to direct heading before moving onto a particular location. This is because I need the drone to move slow, and also turn slow, because I'm taking camera snapshots and framerate is slower if drone moves fast.

However, I was thinking of adding it to core AirSim, instead of using available tools and implementing it myself.

For example, commonly used flight controllers (mostly forks of ardupilot) accept `waypoint`s in mission mode, and they turn drone heading, before approaching the target. Something to that extent.

Anyone is interested ? Or shall I pass ?

vinbo, 2019-08-28:
of course possible, it's up to you to implement it, too simple to add such functionality

msb336, 2019-09-25:
you can control the drone's heading by inputting [`DrivetrainType`](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/types.py#L32) in [moveOnPathAsync](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L291) to `FowardOnly`

closed 2019-09-25 by msb336

YumaTheCompanion, 2019-10-01:
Thanks, this is correct, no need to implement anything special.