microsoft/AirSim #2165: Support for OpenDrive standard?

ThrashAbaddon, 2019-08-28:
I was wondering will `AirSim` support [OpenDrive ](http://www.opendrive.org/)(import or export) or even OpenStreetMap?

ThrashAbaddon, 2019-08-31:
I'd like to mention that there is a project [NETCONVERT](http://sumo.sourceforge.net/userdoc/NETCONVERT.html) which converts between bunch of road network formats, there is a list on the website. This means one can convert maps between open drive and open street map.

adamdarko, 2019-08-31:
yes all of these will work and ive used them in my current build. though with one drive you need to have a vpn enabled (in my case Ive found) to be able to build it properly for unity. unreal engine should be just fine.

closed 2019-09-25 by msb336