microsoft/AirSim #2169: ImageRequest pixels_as_float parameter doesn't work (Unity)

ptotikj, 2019-08-30:
Hi guys,

I am trying two days in a row to find out the problem.
It seems although I am setting `pixels_as_float` to True, my ImageResponse is still `uint8` type.

```python
responses = client.simGetImages([airsim.ImageRequest(camera_name=1, image_type=airsim.ImageType.DepthPerspective, pixels_as_float=True, compress=False)])
```
Here is the response that I have saved in a .txt file:

```json
<ImageResponse> {
    'camera_name': '1',
    'camera_orientation': <Quaternionr> {   'w_val': 0.9937143921852112,
    'x_val': 0.0016330804210156202,
    'y_val': 2.0414589016581886e-05,
    'z_val': 0.1119329035282135},
    'camera_position': <Vector3r> {   'x_val': -219.2003173828125,
    'y_val': 188.4754180908203,
    'z_val': -342.4938049316406},
    'compress': False,
    'height': 256,
    'image_data_float': [   0.0],
    'image_data_uint8': b'!"&\xff'
                        b'\x1f %\xff'
                        b'\x1b\x1c \xff'
                        b'\x1b\x1c \xff'
                            ..........................
                           ............................
                        b'\xb9\xc2\xdd\xff'
                        b'\xb7\xbf\xdb\xff'
                        b'\xb4\xbd\xd8\xff'
                        b'\xb3\xbc\xd7\xff'
                        b'\xb4\xbc\xd8\xff',
    'image_type': 2,
    'message': 'success',
    'pixels_as_float': False,
    'time_stamp': 1567163881748,
    'width': 384}
```

As you can see neither `pixels_as_float`, nor `image_data_float` are populated.

rallen10, 2020-07-22:
I am facing the exact same problem. Did you ever find a solution?

ClarkZha, 2021-02-27:
I bumped into related image request problem and found this issue page.
It seems that one of the latest update is related to you problem. 
Check pull request [#3163].
(Don't know if this is too late as the problem seems to be half a year ago. Hopefully this comment will help developers visiting this page in future.)

closed 2021-10-19 by jonyMarino