microsoft/AirSim #2170: "Error: Blocks could not be compiled. Try rebuilding from source manually" after modifying drone code

vinbo, 2019-08-30:
before I modified the code today, everything goes fine. And I modified a single line in HelloDrone/main.cpp, and build.sh in airsim and succeed. Then when I run the UE4Editor, it's automatic rebuilding went wrong. the error was still the  "Error: Blocks could not be compiled. Try rebuilding from source manually" dialogue box, and error info:
...
Performing 5 actions (2 in parallel)
[1/5] Compile Module.AirSim.cpp
[2/5] Compile Module.AirSim.gen.cpp
fatal error: file '/usr/include/linux/version.h' has been modified since the precompiled header
      '/home/n_li/git/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/n_li/git/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
fatal error: file '/usr/include/linux/version.h' has been modified since the precompiled header
      '/home/n_li/git/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/n_li/git/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
1 error generated.
[3/5] Compile Blocks.cpp
fatal error: file '/usr/include/linux/version.h' has been modified since the precompiled header
      '/home/n_li/git/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/n_li/git/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
1 error generated.
1 error generated.
ERROR: UBT ERROR: Failed to produce item: /home/n_li/git/AirSim_v1.1.10/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
Total build time: 109.41 seconds (Local executor: 0.00 seconds)
LogInit: Warning: Still incompatible or missing module: libUE4Editor-Blocks.so
LogInit: Warning: Still incompatible or missing module: libUE4Editor-AirSim.so
LogExit: Preparing to exit.
...
LogExit: Exiting.
LogInit: Tearing down SDL.
Exiting abnormally (error code: 1)

then what can I do, please? Thanks. (I'm under ubuntu16.04, Airsim v1.1.10)

rajat2004, 2019-08-31:
@vinbo Please have a look at the FAQs, especially the first one, should fix the problem
https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md#faqs

vinbo, 2019-08-31:
> @vinbo Please have a look at the FAQs, especially the first one, should fix the problem
> https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md#faqs

Thank you, it works. I did not find it coz this FAQ is in master branch, not in v1.1.10 branch.

closed 2019-08-31 by vinbo