microsoft/AirSim #2175: airsim.ImageRequest("0", airsim.ImageType.Segmentation, False, False)]) down with time_stamp

vxgu86, 2019-09-02:
#get segmentation image in various formats
responses = client.simGetImages([
    airsim.ImageRequest("0", airsim.ImageType.Segmentation, True), #depth in perspective projection
    airsim.ImageRequest("0", airsim.ImageType.Segmentation, False, False)])  
#scene vision image in uncompressed RGBA array

when the program ran here, the UE4Editor.exe triggerd a breakpoint and quit.
what is the problem?

vxgu86, 2019-09-02:
the breakpoint line is :
            results_[i]->time_stamp = msr::airlib::ClockFactory::get()->nowNanos();
in E:\proj\Plugins\AirSim\Source\RenderRequest.cpp

msb336, 2019-09-03:
Did you manage to get an error raised when VS triggered the breakpoint? If so, can you please post the error to this thread?

ironclownfish, 2019-10-08:
Please re-open if this is still an issue. Closing due to inactivity for now.
If re-opened, please share any error that shows up in VS when the breakpoint is hit.

closed 2019-10-08 by ironclownfish