microsoft/AirSim #2191: Build error due to Airsim setting

02yogesh16, 2019-09-13:
Hi,

I am trying to build Airsim using VisualStudio 2017 on Windos 10. There I am getting the following error, please help me out of this. I have attached a screenshot of the error messages.

[image]

Following are the errors I am getting:

** Warnings

"D:\Softwares\AirSim\AirSim.sln" (default target) (1) ->
"D:\Softwares\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"D:\Softwares\AirSim\AirLib\AirLib.vcxproj" (default target) (3) ->
(ClCompile target) ->
  d:\softwares\airsim\airlib\include\common\airsimsettings.hpp(992): warning C4244: '=': conversion from 'int' to 'uint
16_t', possible loss of data [D:\Softwares\AirSim\AirLib\AirLib.vcxproj]
  d:\softwares\airsim\airlib\include\common\airsimsettings.hpp(992): warning C4244: '=': conversion from 'int' to 'uint
16_t', possible loss of data [D:\Softwares\AirSim\AirLib\AirLib.vcxproj]
  d:\softwares\airsim\airlib\include\common\airsimsettings.hpp(992): warning C4244: '=': conversion from 'int' to 'uint
16_t', possible loss of data [D:\Softwares\AirSim\AirLib\AirLib.vcxproj]
  d:\softwares\airsim\airlib\include\common\airsimsettings.hpp(992): warning C4244: '=': conversion from 'int' to 'uint
16_t', possible loss of data [D:\Softwares\AirSim\AirLib\AirLib.vcxproj]

*Errors:

"D:\Softwares\AirSim\AirSim.sln" (default target) (1) ->
"D:\Softwares\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"D:\Softwares\AirSim\AirLib\AirLib.vcxproj" (default target) (3) ->
(ClCompile target) ->
  d:\softwares\airsim\airlib\include\common\airsimsettings.hpp(992): error C2220: warning treated as error - no 'object
' file generated [D:\Softwares\AirSim\AirLib\AirLib.vcxproj]
  d:\softwares\airsim\airlib\include\common\airsimsettings.hpp(992): error C2220: warning treated as error - no 'object
' file generated [D:\Softwares\AirSim\AirLib\AirLib.vcxproj]
  d:\softwares\airsim\airlib\include\common\airsimsettings.hpp(992): error C2220: warning treated as error - no 'object
' file generated [D:\Softwares\AirSim\AirLib\AirLib.vcxproj]
  d:\softwares\airsim\airlib\include\common\airsimsettings.hpp(992): error C2220: warning treated as error - no 'object
' file generated [D:\Softwares\AirSim\AirLib\AirLib.vcxproj]

    4 Warning(s)
    4 Error(s)

Time Elapsed 00:00:39.22

Sohojoe, 2019-09-13:
@02yogesh16 see temp workaround / commit that builds: https://github.com/microsoft/AirSim/issues/2190

msb336, 2019-09-13:
https://github.com/microsoft/AirSim/pull/2193 this PR should allow you to build successfully again

closed 2019-09-13 by msb336

02yogesh16, 2019-09-16:
Thanks Sohojoe and madratman.