microsoft/AirSim #2192: Signal 11 caught when running Unreal Engine v4.18 on Ubuntu 16.04.5 LTS

BolunDai0216, 2019-09-13:
Hi,

After building Unreal Engine and Airsim I tried to run the UE4Editor and got this error:

Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
Malloc Size=36292 LargeMemoryPoolOffset=232964 
Segmentation fault (core dumped)

can anyone help me understand what this means and what would be the fix.

I am running Unreal Engine v4.18 on Ubuntu 16.04.5 LTS.

sbyebss, 2020-06-17:
I encounter the same problem when inside docker. Does anyone have some solutions to this problem?

nautilor, 2020-07-11:
same on `4.22.1-0+++UE4+Release-4.22`

jonyMarino, 2020-08-19:
AirSim now supports UE v4.24 please try with that version. If the problem persist feel free to open a new issue

closed 2020-08-19 by jonyMarino