microsoft/AirSim #2194: could not get two types image from camera 3

vxgu86, 2019-09-16:
responses = client.simGetImages([
        airsim.ImageRequest("3", airsim.ImageType.Scene),
        airsim.ImageRequest("3", airsim.ImageType.Segmentation)]) 

just like above, images could be got from 0 and 0 , 0 and 3, but not 3 and 3, what is the problem?

vxgu86, 2019-09-17:
I ran python segmentation.py and get the following error infos

ERROR:tornado.application:Uncaught exception, closing connection.
Traceback (most recent call last):
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\tornado\iostream.py", line 562, in wrapper
    return callback(*args)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\tornado\stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 23, in on_read
    for message in self._unpacker:
  File "msgpack\_unpacker.pyx", line 562, in msgpack._cmsgpack.Unpacker.__next__
  File "msgpack\_unpacker.pyx", line 493, in msgpack._cmsgpack.Unpacker._unpack
ValueError: 1970896 exceeds max_bin_len(1048576)
ERROR:tornado.application:Exception in callback functools.partial(<function wrap.<locals>.null_wrapper at 0x000002129D6CB6A8>)
Traceback (most recent call last):
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\tornado\ioloop.py", line 605, in _run_callback
    ret = callback()
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\tornado\stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\tornado\iostream.py", line 562, in wrapper
    return callback(*args)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\tornado\stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\lib\site-packages\msgpackrpc\transport\tcp.py", line 23, in on_read
    for message in self._unpacker:
  File "msgpack\_unpacker.pyx", line 562, in msgpack._cmsgpack.Unpacker.__next__
  File "msgpack\_unpacker.pyx", line 493, in msgpack._cmsgpack.Unpacker._unpack
ValueError: 1970896 exceeds max_bin_len(1048576)

just do not know why

ironclownfish, 2019-10-08:
Can you please post your settings.json file?

saihv, 2020-01-20:
Closing due to inactivity.

closed 2020-01-20 by saihv