microsoft/AirSim #2195: AirSim SITL gps not starting, can't get home position

SusejLav, 2019-09-17:
Hello,

I'm trying to get SITL to work with AirSim + QGC + PX4 but when i start px4 and the sim no matter which command i do the drone behaves incorrectly. Reading through the issues i can only identify that my problem may be with the gps in px4.
When started, i never get the home position log message and can't check it with the 'pos' command, only 'commander status'.
I'm using Windows 10, Airsim 1.2.0, PX4 1.8.2 and QGroundcontrol v3.4.3

[image]

Jaeyoung-Lim, 2019-09-20:
I am using Linux 18.04, but the behavior is correct for me. Have you tried updating Airsim to a more recent version?

SusejLav, 2019-10-09:
At the end it was due to specs, when running in another machine everything went smooth.

closed 2019-10-09 by SusejLav