microsoft/AirSim #2196: the cpp version HelloDrone always trigger breakpoints in airsim RenderRequest::ExecuteTask()

vxgu86, 2019-09-17:
I want to simGetImages(request) with python ,but without success in https://github.com/microsoft/AirSim/issues/2194, so I want to try the cpp version, the environment always trigger breakpoint as follows:

void RenderRequest::ExecuteTask()
{
    if (params_ != nullptr && req_size_ > 0)
    {
        for (unsigned int i = 0; i < req_size_; ++i) {
            FRHICommandListImmediate& RHICmdList = GetImmediateCommandList_ForRenderCommand();
            auto rt_resource = params_[i]->render_target->GetRenderTargetResource();
            if (rt_resource != nullptr) {
                const FTexture2DRHIRef& rhi_texture = rt_resource->GetRenderTargetTexture();
                FIntPoint size;
                auto flags = setupRenderResource(rt_resource, params_[i].get(), results_[i].get(), size);

                //should we be using ENQUEUE_UNIQUE_RENDER_COMMAND_ONEPARAMETER which was in original commit by @saihv
                //https://github.com/Microsoft/AirSim/pull/162/commits/63e80c43812300a8570b04ed42714a3f6949e63f#diff-56b790f9394f7ca1949ddbb320d8456fR64
                if (!params_[i]->pixels_as_float) {
                    //below is undocumented method that avoids flushing, but it seems to segfault every 2000 or so calls
                    RHICmdList.ReadSurfaceData(
                        rhi_texture,
                        FIntRect(0, 0, size.X, size.Y),
                        results_[i]->bmp,
                        flags);
                }
                else {
                    RHICmdList.ReadSurfaceFloatData(
                        rhi_texture,
                        FIntRect(0, 0, size.X, size.Y),
                        results_[i]->bmp_float,
                        CubeFace_PosX, 0, 0
                    );
                }
            }

            results_[i]->time_stamp = msr::airlib::ClockFactory::get()->nowNanos(); <====this place

it said: UE4Editor.exe 已触发了一个断点。

saihv, 2019-10-09:
Could you please post a screenshot of how the error looks like when it breaks there? It's hard to tell which line is the problematic one.

saihv, 2020-01-20:
Closing due to inactivity.

closed 2020-01-20 by saihv