microsoft/AirSim #2198: PythonClient AirSim QuaternionR class error with inverse function

WouterJansen, 2019-09-18:
reproduce:
 - Use the PythonClient folder scripts
 - Create a `Quaternionr` from the AirSim Python library. Defined in _types.py_
 - Try to invert the quaternion with `.inverse()`

example code:
```python
import airsim
import setup_path 

if __name__ == '__main__':
    # connect to the AirSim simulator 
    client = airsim.CarClient()
    client.confirmConnection()

    # get state of the car
    car_state = client.getCarState()

    # get position and orientation
    pos = car_state.kinematics_estimated.position
    orientation = car_state.kinematics_estimated.orientation.inverse()
```
error message:
`pythonTypeError: unsupported operand type(s) for /: 'instance' and 'float'`

bugged code:
_/PythonClient/airsim/types.py_ line 183:
```python
    def inverse(self):
        return self.star() / self.dot(self)
```

Can be fixed by using numpy array in between:
```python
    def inverse(self):
        return self.from_numpy_array(self.star().to_numpy_array() / self.dot(self))

    def from_numpy_array(self, array):
        return Quaternionr( array[0], array[1], array[2], array[3] )
```

madratman, 2019-09-18:
Are you on python2.7?

WouterJansen, 2019-09-19:
yes, as I (and as the [tutorial](https://microsoft.github.io/AirSim/docs/ros/) describes) use ROS Kinetic which is based on Python 2.7.