microsoft/AirSim #2199: AirSim on Unity failing compilation

reneschulte, 2019-09-18:
I was trying to build AirSim for Unity following the steps in the Readme.md. 
Building AirSim with build.cmd worked well with the x64 VS 2017 cmd prompt but running Unity\build,cmd in the same cmd prompt failed. 🤔

It seems like the code is not really compatible with latest from master. I'm using 4414c8e37f1560bbe88dab56766d3827b52640d3 

Here's the full output:

```
D:\Git\OpenSource\Microsoft\AirSim\Unity>build.cmd

D:\Git\OpenSource\Microsoft\AirSim\Unity>REM //---------- copy binaries and include for MavLinkCom in deps ----------

D:\Git\OpenSource\Microsoft\AirSim\Unity>msbuild AirLibWrapper\AirsimWrapper.sln  /target:Clean /target:Build  /property:Configuration=Release /property:Platform=x64
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
Build started 18.09.2019 16:55:06.
Project "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" on node 1 (Clean;Build target(s)).
ValidateSolutionConfiguration:
  Building solution configuration "Release|x64".
Project "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (1) is building "D:\Git\OpenSource\M
icrosoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2) on node 1 (Clean target(s)).
Project "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2) is bu
ilding "D:\Git\OpenSource\Microsoft\AirSim\AirLib\AirLib.vcxproj" (3) on node 1 (Clean target(s)).
_PrepareForClean:
  Deleting file "D:\Git\OpenSource\Microsoft\AirSim\AirLib\temp\AirLib\x64\Release\AirLib.tlog\AirLib.lastbuildstate".
Project "D:\Git\OpenSource\Microsoft\AirSim\AirLib\AirLib.vcxproj" (3) is building "D:\Git\OpenSource\Microsoft\AirSim\
MavLinkCom\MavLinkCom.vcxproj" (4) on node 1 (Clean target(s)).
_PrepareForClean:
  Deleting file "D:\Git\OpenSource\Microsoft\AirSim\MavLinkCom\temp\MavLinkCom\x64\Debug\MavLinkCom.tlog\MavLinkCom.las
  tbuildstate".
Done Building Project "D:\Git\OpenSource\Microsoft\AirSim\MavLinkCom\MavLinkCom.vcxproj" (Clean target(s)).

Done Building Project "D:\Git\OpenSource\Microsoft\AirSim\AirLib\AirLib.vcxproj" (Clean target(s)).

Project "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2) is bu
ilding "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj" (5) on node 1 (Clea
n target(s)).
_PrepareForClean:
  Deleting file "x64\Release\AirsimWrapper.tlog\AirsimWrapper.lastbuildstate".
Done Building Project "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj" (Cle
an target(s)).

Done Building Project "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metap
roj" (Clean target(s)).

Project "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (1) is building "D:\Git\OpenSource\M
icrosoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2:2) on node 1 (default targets).
Project "D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2:2) is
building "D:\Git\OpenSource\Microsoft\AirSim\AirLib\AirLib.vcxproj" (3:4) on node 1 (default targets).
Project "D:\Git\OpenSource\Microsoft\AirSim\AirLib\AirLib.vcxproj" (3:4) is building "D:\Git\OpenSource\Microsoft\AirSi
m\MavLinkCom\MavLinkCom.vcxproj" (4:2) on node 1 (default targets).
InitializeBuildStatus:
  Creating "D:\Git\OpenSource\Microsoft\AirSim\MavLinkCom\temp\MavLinkCom\x64\Debug\MavLinkCom.tlog\unsuccessfulbuild"
  because "AlwaysCreate" was specified.
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\CL.exe /c /I
  common_utils /Iinclude /ZI /JMC /nologo /W3 /WX /diagnostics:classic /Od /D _WIN32 /D _WIN32_WINNT=0x0600 /D NOMINMAX
   /D VC_EXTRALEAN /D _SCL_SECURE_NO_WARNINGS /D _CRT_SECURE_NO_WARNINGS /D _DEBUG /D _LIB /D _UNICODE /D UNICODE /Gm-
  /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"D:\Git\OpenSource\Microsoft\AirSim\MavLinkCo
  m\temp\MavLinkCom\x64\Debug\\" /Fd"D:\Git\OpenSource\Microsoft\AirSim\MavLinkCom\temp\MavLinkCom\x64\Debug\MavLinkCom
  .pdb" /FR"D:\Git\OpenSource\Microsoft\AirSim\MavLinkCom\temp\MavLinkCom\x64\Debug\\" /Gd /TP /FC /errorReport:queue c
  ommon_utils\FileSystem.cpp common_utils\ThreadUtils.cpp src\UdpSocket.cpp src\impl\UdpSocketImpl.cpp src\AdHocConnect
  ion.cpp src\impl\AdHocConnectionImpl.cpp src\impl\MavLinkFtpClientImpl.cpp src\impl\MavLinkNodeImpl.cpp src\impl\MavL
  inkTcpServerImpl.cpp src\impl\onecore\OneCoreFindSerialPorts.cpp src\impl\windows\WindowsFindSerialPorts.cpp src\MavL
  inkFtpClient.cpp src\MavLinkLog.cpp src\MavLinkMessageBase.cpp src\MavLinkMessages.cpp src\MavLinkNode.cpp src\Semaph
  ore.cpp src\MavLinkTcpServer.cpp src\MavLinkVehicle.cpp src\impl\MavLinkVehicleImpl.cpp src\MavLinkConnection.cpp src
  \impl\MavLinkConnectionImpl.cpp src\MavLinkVideoStream.cpp src\impl\MavLinkVideoStreamImpl.cpp src\serial_com\SerialP
  ort.cpp src\serial_com\SocketInit.cpp src\serial_com\TcpClientPort.cpp src\serial_com\UdpClientPort.cpp src\serial_co
[... 235 lines cut ...]
Api::resetImplementation': method with override specifier 'override' did not override any base class methods [D:\Git\Op
enSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj]
  d:\git\opensource\microsoft\airsim\unity\airlibwrapper\airsimwrapper\source\simmode\simmodebase.cpp(18): error C2259:
 'WorldSimApi': cannot instantiate abstract class [D:\Git\OpenSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper
\AirsimWrapper.vcxproj]
  d:\git\opensource\microsoft\airsim\unity\airlibwrapper\airsimwrapper\source\worldsimapi.h(16): error C3668: 'WorldSim
Api::resetImplementation': method with override specifier 'override' did not override any base class methods [D:\Git\Op
enSource\Microsoft\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj]

    0 Warning(s)
    3 Error(s)

Time Elapsed 00:01:57.94

D:\Git\OpenSource\Microsoft\AirSim\Unity>if ERRORLEVEL 1 goto :buildfailed

D:\Git\OpenSource\Microsoft\AirSim\Unity>echo(

D:\Git\OpenSource\Microsoft\AirSim\Unity>echo #### Build failed - see messages above.  1>&2
#### Build failed - see messages above.

D:\Git\OpenSource\Microsoft\AirSim\Unity>exit /b 1

D:\Git\OpenSource\Microsoft\AirSim\Unity>
```

rajat2004, 2019-09-18:
@reneschulte Thanks for reporting the problem!
I have opened https://github.com/microsoft/AirSim/pull/2200 for fixing this, but tested till compilation only since don't have Unity installed right now
It would be great if you could test it on your system also

reneschulte, 2019-09-19:
That fixes it! Thank you for the quick fix! 🙌

closed 2019-09-19 by reneschulte