microsoft/AirSim #2201: moveOnPathAsync or Moveonpath did not work at all ,but moveByVelocityZAsync works well

vxgu86, 2019-09-19:
the code are as follows:
		vector<Vector3r> path = { Vector3r(0, -20, -2), Vector3r(100, -20, high) };
               client.moveOnPathAsync(path, speed, duration, driveTrain, yaw_mode);
		std::this_thread::sleep_for(std::chrono::duration<double>(duration));

or droneshell like this:
Moveonpath -path 0,-256,-4,126,-256,-4,126,0,-4,-5,0,-4 -velocity 10 -lookahead 10

not work , but moveByVelocityZAsync works well, what is the problem?

vxgu86, 2019-09-21:
client.moveToPositionAsync(0, 12, high, speed, 20, driveTrain, yaw_mode);
		std::this_thread::sleep_for(std::chrono::duration<double>(20));

didnot work either.....

saihv, 2019-10-09:
Does the drone not move at all?

saihv, 2020-01-20:
Closing due to inactivity.

closed 2020-01-20 by saihv