microsoft/AirSim #2205: DQNdrone.py

oooohhhhhhh, 2019-09-22:
When i run this code on spyder , it says No module named 'setup_path' , i wonder what "setup_path" meaning in this code

appreciate

msb336, 2019-09-23:
setup path is included in both the `PythonClient/multirotor` and `PythonClient/car` folders. [here is a link](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/setup_path.py)

`import setup_path` is the first module to be imported in all of our PythonClient scripts, as it ensures that the airsim python module to be loaded is the version defined in the repo.

If you installed airsim globally (`pip install airsim`), you can remove this line:
```
import setup_path
```
and your script should run without issue.

oooohhhhhhh, 2019-09-24:
thank you very much

closed 2019-09-24 by oooohhhhhhh