microsoft/AirSim #2209: waitOnLastTask crashing

dimikout3, 2019-09-24:
I am trying to use the waitOnLastTask() function. I am doing something like:

client = airsim.MultirotorClient()
client.moveToPositionAsync(x,y,z,speed,vehicle_name=droneName).join()
client.waitOnLastTask()

but I get the following error: 
 File "C:\Users\DKoutras\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\airsim\client.py", line 199, in waitOnLastTask
    return self.client.call('waitOnLastTask', timeout_sec)
NameError: name 'self' is not defined

I tried to add add the self argument ( def waitOnLastTask(self, timeout_sec = float('nan')): ...), but it alse failed.

Any ideas ?

Francesco-Rose, 2019-09-28:
Did you find any solution for this problem? I am encountering the same issue.

Thank you

dimikout3, 2019-09-28:
Unfortunately not yet ... I will come back on it tommorow probably. I am using the time.sleep() for now, though it is not an ideal fix ...

Francesco-Rose, 2019-09-28:
I tried it too, but I do not know how much time I have to wait. I should create functions for my purpose to evaluate the time needed but it is a mess. If you know how to fix the problem with this API let me know.

rajat2004, 2019-09-28:
I'm not sure about this but have you tried without using the `waitOnLastTask` function in Python?
From [this](https://github.com/microsoft/AirSim/blob/master/docs/apis.md#async-methods-duration-and-max_wait_seconds), it doesn't seem to be for Python, only for C++ API

From that page, using the `.join()` call should be enough, It's also mentioned as a comment in the `Hello Drone` Python exmaple
`# Async methods returns Future. Call join() to wait for task to complete.`

jpierman7, 2019-09-28:
> and it doesn't seem to be for Python, only for C++ APIc++a

unnamed.jpg
<https://drive.google.com/file/d/11fTV5CTh5EapEM0NzMqP6s8hLRVRS4o01w/view?usp=drivesdk>

Francesco-Rose, 2019-09-29:
> I'm not sure about this but have you tried without using the `waitOnLastTask` function in Python?
> From [...]

My problem is that I have the API inside a for loop, so python skips the first points and goes directly to the last command. So I need to put something as time.sleep() to stop python until the drone finishes his path. Here is the problem, each path requires a different time to be accomplished and so each iteration has to sleep for different durations. The only way is to construct time functions to evaluate every iteration the time required. 

Anyway, if someone fixes this API, it will be very useful.

dimikout3, 2019-09-29:
Here is how i solve it: I am creating a list of task =[], then inside every loop I am appending a specific task to the list, once the for-loop is finished I am calling the joint() method for all task. Keep in mind that I am working on drone swarm, so I do not know if this solution can apply to your use case:

tasks = []
for drone in DronesList:
  t = client.moveToPositionAsync(x,y,z,vehicle_name=drone)
  tasks.append(t)
for t in tasks:
  t.join()

dimikout3, 2019-09-29:
If I had only one drone and multiple actions (something I believe its similar to your use case), I would do the following

for action in ActionsLists:
__t = action ()
__if idx == len(ActionsLists):
____t.join()

dimikout3, 2019-09-29:
Or even more simple like:

for-loop:
__a1 = client.moveToPositionAsync()
__a2 = client.getImages()
__a2.join()