microsoft/AirSim #2210: The problem of controlling uav

oooohhhhhhh, 2019-09-24:
I'm learning for controlling the UAV.
I try to run  QDNdrone.py  (https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/DQNdrone.py) .
It could fly automatically , but i wonder how to do to make it stop , and doing path planning,
are there any training vedio , or topic to learn about it

appreciate

msb336, 2019-09-24:
To make the drone stop, you can send the following command:
`client.moveByVelocityAsync(0,0,0,1, vehicle_name="{your drone}")`

[Here is a demo script for having a drone move on a path](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/path.py)

In terms of learning how path planning is done, [here is a survey paper to help you get started on your studies](https://ieeexplore.ieee.org/document/7053093).

oooohhhhhhh, 2019-09-24:
Thank you very much