microsoft/AirSim #2214: Failed in loading the car/multirotor mode

ChenChangHaoWS, 2019-09-25:
I'm totally new at AirSim. Can anyone help with this problem?

When I start UE4Editor there will be:
[image]
[image]

And after I load the block.project there will be:
[image]
[image]

Because of the problem I can't load any car or multirotor mode, and ask for some advice...

msb336, 2019-09-25:
Check to see if the airsim plugin is in the project folder (`Blocks/Plugins/AirSim`). If it isn't, copy and paste `Unreal/Plugins/AirSim` into `Blocks/Plugins`, delete any folders titled `Binaries` or `Intermediates` in the blocks project, and try running it again.

ChenChangHaoWS, 2019-09-26:
@

> Check to see if the airsim plugin is in the project folder (`Blocks/Plugins/AirSim`). If it isn't, copy and paste `Unreal/Plugins/AirSim` into `Blocks/Plugins`, delete any folders titled `Binaries` or `Intermediates` in the blocks project, and try running it again.

Thanks for your advice. I check the folder, the 'AirSim.uplugin' file exists. I also tried to delete all 'Binaries' and 'Intermediates' files in the blocks and rebuild, and got the following results:

[image]

msb336, 2019-09-26:
@ChenChangHaoWS can you please post your log file to this thread? `Unreal/Environments/Blocks/Saved/Logs`

ChenChangHaoWS, 2019-09-26:
> @ChenChangHaoWS can you please post your log file to this thread? `Unreal/Environments/Blocks/Saved/Logs`

Sure,  and thank you for your patience. The log is a little bit long, I will send it in two parts:
ubuntu@cch:~/software/UnrealEngine$ Engine/Binaries/Linux/UE4EditorIncreasing per-process limit of core file size to infinity.
LogInit: Display: Running engine without a game
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(304532448) SRandInit(304532448).
LogInit: Read version file with build ID 'f2e9955c-2326-41b5-9681-79219e303d55'
LogInit: Registered custom module enumerator with build ID 'f2e9955c-2326-41b5-9681-79219e303d55'
LogTaskGraph: Started task graph with 5 named threads and 53 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.358374
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaPlayerEditor
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin MobileLauncherProfileWizard
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin OnlineSubsystemIOS
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemGooglePlay
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Plugin shader directory ../../../Engine/Plugins/Runtime/Oculus/OculusVR/Shaders found
LogPluginManager: Mapping shader source directory ../../../Engine/Plugins/Runtime/Oculus/OculusVR/Shaders to virtual directory /Plugin/OculusVR
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin AndroidDeviceProfileSelector
LogPluginManager: Mounting plugin IOSDeviceProfileSelector
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin AndroidMoviePlayer
LogPluginManager: Mounting plugin AppleMoviePlayer
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin GoogleCloudMessaging
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin AppleARKit
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin LevelSequenceEditor
LogPluginManager: Mounting plugin MatineeToLevelSequence
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin HTML5Networking
[... 1055 lines cut ...]
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Editor/SpeedTreeImporter/Binaries/Linux/libUE4Editor-SpeedTreeImporter.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Editor/PluginBrowser/Binaries/Linux/libUE4Editor-PluginBrowser.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Editor/GameplayTagsEditor/Binaries/Linux/libUE4Editor-GameplayTagsEditor.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Editor/FacialAnimation/Binaries/Linux/libUE4Editor-FacialAnimation.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Editor/FacialAnimation/Binaries/Linux/libUE4Editor-FacialAnimationEditor.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Online/OnlineSubsystemNull/Binaries/Linux/libUE4Editor-OnlineSubsystemNull.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Online/OnlineSubsystem/Binaries/Linux/libUE4Editor-OnlineSubsystem.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Online/OnlineSubsystemUtils/Binaries/Linux/libUE4Editor-OnlineSubsystemUtils.so (API version 0).
[... cut ...]
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/ExampleDeviceProfileSelector/Binaries/Linux/libUE4Editor-ExampleDeviceProfileSelector.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/RuntimePhysXCooking/Binaries/Linux/libUE4Editor-RuntimePhysXCooking.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/Steam/SteamVR/Binaries/Linux/libUE4Editor-SteamVR.so (API version 0).

ChenChangHaoWS, 2019-09-26:
> @ChenChangHaoWS can you please post your log file to this thread? `Unreal/Environments/Blocks/Saved/Logs`

The second part which is for rebuild:

ubuntu@cch:~/software/UnrealEngine$ LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/Steam/SteamVR/Binaries/Linux/libUE4Editor-SteamVRController.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Runtime/AndroidPermission/Binaries/Linux/libUE4Editor-AndroidPermission.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/MovieScene/ActorSequence/Binaries/Linux/libUE4Editor-ActorSequence.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/MovieScene/LevelSequenceEditor/Binaries/Linux/libUE4Editor-LevelSequenceEditor.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/MovieScene/MatineeToLevelSequence/Binaries/Linux/libUE4Editor-MatineeToLevelSequence.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Experimental/CharacterAI/Binaries/Linux/libUE4Editor-CharacterAI.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Messaging/UdpMessaging/Binaries/Linux/libUE4Editor-UdpMessaging.so (API version 0).
LogModuleManager: Display: Found up-to-date module file ../../../Engine/Plugins/Messaging/TcpMessaging/Binaries/Linux/libUE4Editor-TcpMessaging.so (API version 0).
LogInit: Warning: Incompatible or missing module: libUE4Editor-Blocks.so
LogInit: Warning: Incompatible or missing module: libUE4Editor-AirSim.so
LogInit: Initializing SDL.
LogInit: Using SDL_WINDOW_OPENGL
LogInit: Initialized SDL 2.0.5 revision: 11184 (hg-11184:04063928c4a8 with EpicExtensions) (compiled against 2.0.5)
LogInit: Display metrics:
LogInit:   PrimaryDisplayWidth: 1440
LogInit:   PrimaryDisplayHeight: 900
LogInit:   PrimaryDisplayWorkAreaRect:
LogInit:     Left=65, Top=24, Right=1440, Bottom=900
LogInit:   VirtualDisplayRect:
LogInit:     Left=65, Top=24, Right=1440, Bottom=900
LogInit:   TitleSafePaddingSize: X=0.000 Y=0.000
LogInit:   ActionSafePaddingSize: X=0.000 Y=0.000
LogInit:   Number of monitors: 1
LogInit:     Monitor 0
LogInit:       Name: Philips 190VL 19"
LogInit:       ID: display0
LogInit:       NativeWidth: 1440
LogInit:       NativeHeight: 900
LogInit:       bIsPrimary: true
STUBBED: FDesktopPlatformLinux::GetNativeFeedbackContext at /home/ubuntu/software/UnrealEngine/Engine/Source/Developer/DesktopPlatform/Private/Linux/DesktopPlatformLinux.cpp:269 (GetNativeFeedbackContext)
Running /home/ubuntu/software/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Linux -project="/home/ubuntu/software/AirSim/Unreal/Environments/Blocks/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE

Running Mono...

~/software/UnrealEngine/Engine ~/software/UnrealEngine
Using 'git status' to determine working set for adaptive non-unity build.
Creating makefile for Blocks (no existing makefile)
Performing full C++ include scan (no include cache file)
Using clang version '3.8.0' (string), 3 (major), 8 (minor), 0 (patch)
@progress push 5%
@progress 'Generating code...' 0%
Building UnrealHeaderTool...
Using 'git status' to determine working set for adaptive non-unity build.
Target is up to date
Total build time: 0.45 seconds (NoActionsToExecute executor: 0.00 seconds)
@progress 'Generating code...' 33%
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/home/ubuntu/software/AirSim/Unreal/Environments/Blocks/Blocks.uproject" "/home/ubuntu/software/AirSim/Unreal/Environments/Blocks/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(-1935847367) SRandInit(-1935847367).
LogTaskGraph: Started task graph with 4 named threads and 20 total threads with 1 sets of task threads.
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaPlayerEditor
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
[... 1622 lines cut ...]
[2019.09.26-15.37.46:527][  0]LogCompile: Success: Parsed 164 line(s), 48 statement(s) in 0.00 secs.
[2019.09.26-15.37.46:537][  0]LogCompile: Generating code for module 'Kismet'
[2019.09.26-15.37.46:539][  0]LogCompile: Success: Parsed 524 line(s), 82 statement(s) in 0.00 secs.
[2019.09.26-15.37.46:541][  0]LogCompile: Generating code for module 'ContentBrowser'
[2019.09.26-15.37.46:543][  0]LogCompile: Success: Parsed 21 line(s), 11 statement(s) in 0.00 secs.
[2019.09.26-15.37.46:692][  0]LogCompile: Generating code for module 'AnimGraph'
[... cut ...]
[2019.09.26-15.37.46:757][  0]LogCompile: Generating code for module 'PIEPreviewDeviceProfileSelector'
[2019.09.26-15.37.46:760][  0]LogCompile: Success: Parsed 129 line(s), 21 statement(s) in 0.00 secs.
[2019.09.26-15.37.47:410][  0]LogCompile: Generating code for module 'UnrealEd'
[2019.09.26-15.37.47:504][  0]LogCompile: Success: Parsed 25572 line(s), 7074 statement(s) in 0.31 secs.
[2019.09.26-15.37.47:508][  0]LogCompile: Generating code for module 'ConfigEditor'
[2019.09.26-15.37.47:512][  0]LogCompile: Success: Parsed 51 line(s), 22 stateme

msb336, 2019-09-26:
I have sad news for you.

```
fatal error: file '/usr/include/linux/version.h' has been modified since the
precompiled header
```

This happens occasionally with unreal in linux. You will need to rebuild unreal in order for you to successfully build this project.

rajat2004, 2019-09-26:
Try deleting the following file and try again
```
'/home/ubuntu/software/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
```
Source - https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md#faqs

ChenChangHaoWS, 2019-09-27:
> Try deleting the following file and try again
> 
> ```
> '/home/ubuntu/software/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
> ```
> 
> Source - https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md#faqs

Thanks for your advice. 
I tried to delet the 'SharedPCH.Engine.h.gch' file and all 'Binaries' & 'Intermediates' files in the '/home/ubuntu/software/AirSim/Unreal/Environments/Blocks' floder and rebuild. It seems to return to my original problem that some files can't be found or can't be loaded:
[image]

Here is my log:
ubuntu@cch:~/software/UnrealEngine$ Engine/Binaries/Linux/UE4Editor
Increasing per-process limit of core file size to infinity.
LogInit: Display: Running engine without a game
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(2001700416) SRandInit(2001700416).
LogInit: Read version file with build ID 'f2e9955c-2326-41b5-9681-79219e303d55'
LogInit: Registered custom module enumerator with build ID 'f2e9955c-2326-41b5-9681-79219e303d55'
LogTaskGraph: Started task graph with 5 named threads and 53 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.051870
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaPlayerEditor
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin MobileLauncherProfileWizard
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin OnlineSubsystemIOS
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemGooglePlay
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Plugin shader directory ../../../Engine/Plugins/Runtime/Oculus/OculusVR/Shaders found
LogPluginManager: Mapping shader source directory ../../../Engine/Plugins/Runtime/Oculus/OculusVR/Shaders to virtual directory /Plugin/OculusVR
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin AndroidDeviceProfileSelector
LogPluginManager: Mounting plugin IOSDeviceProfileSelector
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin AndroidMoviePlayer
LogPluginManager: Mounting plugin AppleMoviePlayer
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
[... 1096 lines cut ...]
LogInit:       Name: Philips 190VL 19"
LogInit:       ID: display0
LogInit:       NativeWidth: 1440
LogInit:       NativeHeight: 900
LogInit:       bIsPrimary: true
[2019.09.27-01.35.18:719][606]LogModuleManager: Shutting down and abandoning module SSL (38)
[2019.09.27-01.35.18:722][606]LogModuleManager: Shutting down and abandoning module AnimationModifiers (34)
[2019.09.27-01.35.18:722][606]LogModuleManager: Shutting down and abandoning module PropertyEditor (33)
[2019.09.27-01.35.18:722][606]LogModuleManager: Shutting down and abandoning module AudioEditor (30)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module TextureCompressor (28)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module ShaderCore (26)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module Landscape (24)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (22)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module OpenGLDrv (20)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (18)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module Renderer (16)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module Engine (14)
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module CoreUObject (12)
[... cut ...]
[2019.09.27-01.35.18:723][606]LogModuleManager: Shutting down and abandoning module CookedIterativeFile (8)
[2019.09.27-01.35.18:724][606]LogModuleManager: Shutting down and abandoning module StreamingFile (6)
[2019.09.27-01.35.18:724][606]LogModuleManager: Shutting down and abandoning module SandboxFile (4)
[2019.09.27-01.35.18:724][606]LogModuleManager: Shutting down and abandoning module PakFile (2)
[2019.09.27-01.35.18:728][606]LogExit: Exiting.
[2019.09.27-01.35.18:728][606]LogInit: Tearing down SDL

ChenChangHaoWS, 2019-09-27:
> Try deleting the following file and try again
> 
> ```
> '/home/ubuntu/software/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
> ```
> 
> Source - https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md#faqs

The second part:

ubuntu@cch:~/software/UnrealEngine$ STUBBED: 

[2019.09.27-01.35.46:467][  0]LogCompile: Success: Parsed 1528 line(s), 787 statement(s) in 0.07 secs.
[2019.09.27-01.35.46:469][  0]LogCompile: Preparsing 172 modules took 0.91 seconds
[2019.09.27-01.35.46:469][  0]LogCompile: Parsing took 4.87 seconds
[2019.09.27-01.35.46:469][  0]LogCompile: Code generation took 3.17 seconds
[2019.09.27-01.35.46:469][  0]LogCompile: ScriptPlugin overhead was 0.00 seconds
[2019.09.27-01.35.46:469][  0]LogCompile: Macroize time was 0.10 seconds
Reflection code generated for BlocksEditor in 9.7676125 seconds
@progress 'Generating code...' 67%
@progress 'Generating code...' 100%
@progress pop
Using bundled libc++ standard C++ library.
Using fast way to relink  circularly dependent libraries (no FixDeps).
Performing 430 actions (31 in parallel)
@progress 'Compiling C++ source code...' 0%
[1/430] Compile SharedPCH.Engine.h
[2/430] Compile Module.SourceControl.cpp
[3/430] Compile Module.SourceControl.gen.cpp
[4/430] Compile Module.Foliage.cpp
[5/430] Compile Module.Foliage.gen.cpp
[6/430] Compile Module.BlueprintGraph.1_of_5.cpp
[7/430] Compile Module.BlueprintGraph.gen.cpp
[8/430] Compile Module.BlueprintGraph.5_of_5.cpp
[9/430] Compile Module.BlueprintGraph.4_of_5.cpp
[10/430] Compile Module.BlueprintGraph.3_of_5.cpp
[11/430] Compile Module.BlueprintGraph.2_of_5.cpp
[12/430] Compile Module.AnimGraphRuntime.gen.cpp
[13/430] Compile Module.AnimGraphRuntime.cpp
[14/430] Compile Module.MovieScene.cpp
[15/430] Compile Module.MovieScene.gen.cpp
[16/430] Compile Module.AnimGraph.cpp
[17/430] Compile Module.AnimGraph.gen.cpp
[18/430] Compile Module.ClothingSystemRuntime.gen.cpp
[19/430] Compile Module.ClothingSystemRuntime.cpp
[20/430] Compile Module.MovieSceneTracks.gen.cpp
[21/430] Compile Module.MovieSceneTracks.cpp
[22/430] Compile Module.LevelSequence.gen.cpp
[23/430] Compile Module.LevelSequence.cpp
[24/430] Compile Module.FunctionalTesting.gen.cpp
[25/430] Compile Module.FunctionalTesting.cpp
[26/430] Compile Module.HierarchicalLODOutliner.cpp
[27/430] Compile Module.PixelInspectorModule.cpp
[28/430] Compile Module.PixelInspectorModule.gen.cpp
[29/430] Compile Module.HierarchicalLODOutliner.gen.cpp
[30/430] Compile Module.Renderer.gen.cpp
[31/430] Compile Module.Renderer.1_of_11.cpp
[32/430] Compile Module.GameplayTags.gen.cpp
[33/430] Compile Module.GameplayTags.cpp
@progress 'Compiling C++ source code...' 1%
[34/430] Compile Module.CinematicCamera.gen.cpp
[35/430] Compile Module.Landscape.2_of_3.cpp
[36/430] Compile Module.Landscape.3_of_3.cpp
[37/430] Compile Module.Landscape.gen.cpp
@progress 'Compiling C++ source code...' 2%
[38/430] Compile Module.CollisionAnalyzer.cpp
[39/430] Compile Module.CinematicCamera.cpp
[40/430] Compile Module.Landscape.1_of_3.cpp
[41/430] Compile Module.UMG.gen.6_of_6.cpp
@progress 'Compiling C++ source code...' 3%
[... 1323 lines cut ...]
[2019.09.27-01.40.00:067][  0]LogContentBrowser: Native class hierarchy populated in 0.0305 seconds. Added 2363 classes and 503 folders.
[2019.09.27-01.40.00:070][  0]LogCollectionManager: Rebuilt the GUID cache for 0 collections in 0.000001 seconds
[2019.09.27-01.40.00:211][  0]LogContentBrowser: Native class hierarchy updated for 'WidgetCarousel' in 0.0005 seconds. Added 0 classes and 0 folders.
[2019.09.27-01.40.00:302][  0]LogContentBrowser: Native class hierarchy updated for 'AddContentDialog' in 0.0007 seconds. Added 0 classes and 0 folders.
[2019.09.27-01.40.00:498][  0]LogContentBrowser: Native class hierarchy updated for 'HierarchicalLODOutliner' in 0.0008 seconds. Added 1 classes and 2 folders.
[2019.09.27-01.40.00:498][  0]LogLoad: (Engine Initialization) Total time: 284.15 seconds
[2019.09.27-01.40.00:498][  0]LogLoad: (Engine Initialization) Total Blueprint compile time: 0.00 seconds
[2019.09.27-01.40.00:498][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2019.09.27-01.40.00:511][  0]LogSlate: Took 0.000358 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2019.09.27-01.40.00:515][  0]LogSlate: Took 0.001318 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/DroidSansFallback.ttf' (3848K)
[2019.09.27-01.40.00:516][  0]LogSlate: Took 0.000169 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Bold.ttf' (160K)
[2019.09.27-01.40.00:524][  0]LogSlate: Took 0.000845 seconds to synchronously load lazily loaded font '../../../Engine/Content/Editor/Slate/Fonts/FontAwesome.ttf' (139K)
[2019.09.27-01.40.00:762][  0]LogRenderer: Reallocating scene render targets to support 864x544 Format 10 NumSamples 1 (Frame:1).
[2019.09.27-01.40.00:932][  0]LogInit: Adding external input plugin.
[2019.09.27-01.40.00:951][  0]LogContentStreaming: Texture pool size now 1000 MB
[2019.09.27-01.40.00:953][  0]LogLinux: Scale at X=343.000000, Y=290.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
[... cut ...]
[2019.09.27-01.40.01:369][  2]LogCollectionManager: Rebuilt the object cache for 0 collections in 0.000000 seconds (found 0 objects)
[2019.09.27-01.40.01:369][  2]LogCollectionManager: Fixed up redirectors for 0 collections in 0.000063 seconds (updated 0 objects)
[2019.09.27-01.40.23:396][623]LogLinux: Scale at X=720.000000, Y=450.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
[2019.09.27-01.40.23:556][627]LogLinux: Setting swap interval to 'Immediate'

rajat2004, 2019-09-27:
Next option would be to run the `clean_rebuild.sh` script from the AirSim directory. It'll clean out everything, download the dependencies again and rebuild. Then follow the steps as usual
If nothing else works, then maybe try removing AirSim, Unreal Engine & start again from scratch

ChenChangHaoWS, 2019-09-27:
> Next option would be to run the `clean_rebuild.sh` script from the AirSim directory. It'll clean out everything, download the dependencies again and rebuild. Then follow the steps as usual
> If nothing else works, then maybe try removing AirSim, Unreal Engine & start again from scratch

Thanks for your help, everthing works fine normally after run the clean_rebuild.sh script except some 'can't find files' warning. Is that ok or should I reinstall Unreal Engine , here are the warnings:

[2019.09.27-09.19.36:920][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.09.27-09.19.36:920][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.09.27-09.19.36:920][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:920][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:921][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:921][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:921][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.09.27-09.19.36:921][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.09.27-09.19.36:921][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:921][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:922][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:922][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:922][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.09.27-09.19.36:922][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.09.27-09.19.36:922][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:922][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:923][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:923][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:923][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.09.27-09.19.36:923][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.09.27-09.19.36:923][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:923][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:924][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:924][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:927][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png' error.
[2019.09.27-09.19.36:927][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png
[2019.09.27-09.19.36:928][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2019.09.27-09.19.36:928][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[2019.09.27-09.19.36:928][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2019.09.27-09.19.36:928][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[2019.09.27-09.19.36:928][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png' error.
[2019.09.27-09.19.36:928][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png
[... cut ...]
[2019.09.27-09.19.48:434][  0]LogInit: Transaction tracking system initialized
[2019.09.27-09.19.48:889][  0]BlueprintLog: New page: 编辑器加载
[2019.09.27-09.19.50:427][  0]LocalizationService: 本地化服务已被禁用
1001[2019.09.27-09.19.51:020][  0]LogLinker: Warning: 无法寻找到文件'/Game/FlyingBP/FlyingOverview'
1001[2019.09.27-09.19.51:021][  0]LogLinker: Warning: 无法寻找到文件'/Game/FlyingBP/FlyingOverview'
[2019.09.27-09.19.51:021][  0]LogUObjectGlobals: Warning: 寻找对象'Class /Game/FlyingBP/FlyingOverview.FlyingOverview_C'失败

ChenChangHaoWS, 2019-09-27:
> Next option would be to run the `clean_rebuild.sh` script from the AirSim directory. It'll clean out everything, download the dependencies again and rebuild. Then follow the steps as usual
> If nothing else works, then maybe try removing AirSim, Unreal Engine & start again from scratch

Thanks for your help, everthing works normally now after run the clean_rebuild.sh script except some 'can't find files' warnings. Is that ok or should I reinstall Unreal Engine to avoid possible errors?  Here are the warnings:

[2019.09.27-09.19.36:920][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.09.27-09.19.36:920][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.09.27-09.19.36:920][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:920][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:921][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:921][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:921][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.09.27-09.19.36:921][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.09.27-09.19.36:921][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:921][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:922][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:922][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:922][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.09.27-09.19.36:922][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.09.27-09.19.36:922][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:922][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:923][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:923][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:923][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.09.27-09.19.36:923][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.09.27-09.19.36:923][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:923][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:924][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.09.27-09.19.36:924][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.09.27-09.19.36:927][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png' error.
[2019.09.27-09.19.36:927][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png
[2019.09.27-09.19.36:928][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2019.09.27-09.19.36:928][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[2019.09.27-09.19.36:928][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png' error.
[2019.09.27-09.19.36:928][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_128x.png
[2019.09.27-09.19.36:928][ 0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png' error.
[2019.09.27-09.19.36:928][ 0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png
[... cut ...]
[2019.09.27-09.19.47:890][ 0]LogEngineSessionManager: EngineSessionManager initialized
[2019.09.27-09.19.48:434][ 0]LogInit: Transaction tracking system initialized
[2019.09.27-09.19.48:889][ 0]BlueprintLog: New page: 编辑器加载
[2019.09.27-09.19.50:427][ 0]LocalizationService: 本地化服务已被禁用
1001[2019.09.27-09.19.51:020][ 0]LogLinker: Warning: 无法寻找到文件'/Game/FlyingBP/FlyingOverview'
1001[2019.09.27-09.19.51:021][ 0]LogLinker: Warning: 无法寻找到文件'/Game/FlyingBP/FlyingOverview'
[2019.09.27-09.19.51:021][ 0]LogUObjectGlobals: Warning: 寻找对象'Class /Game/FlyingBP/FlyingOverview.FlyingOverview_C'失败

rajat2004, 2019-09-27:
Great!
Even I get these "can't find files" warnings, but since they are images and seem to be icons or logos, I ignore them

closed 2019-09-27 by msb336