microsoft/AirSim #2217: Select "Generate Visual Studio Project Files" ,but do not generate ".sln" files

sbkybhy, 2019-09-27:
Follow the tutorial，I installed "Landscape Mountains", created "New C++ class",and copy “Unreal\Plugins”,too.I also changed the "LandscapeMountains uproject". However, after executing "Generate Visual Studio Project Files", the .sln file will not be generated. Using the "Unreal Editor" open "LandscapeMountains. Uproject", it will display what is shown in the figure below. How to solve this problem?
[image]
[image]
[image]

msb336, 2019-09-27:
The log from attempting to open your project will provide more information as to why this failed. Can you add it to this thread?
It is located here: `{your project}/Saved/Logs/`

sbkybhy, 2019-09-28:
The file in the attachment is the latest log record. Thank you so much！

Re-Ell, 2019-10-01:
I was getting a similar issue but couldn't find a proper solution. I tried building with all other versions but I doesn't seem to solve the error.

bmalhigh, 2019-10-01:
I am not seeing the any file attachment here. Where is the log file?

BTW, can you try running the explorer window as an Administrator and then retry?

sbkybhy, 2019-10-03:
[TaiYangShenMiao.log](https://github.com/microsoft/AirSim/files/3684292/TaiYangShenMiao.log)
[TaiYangShenMiao-backup-2019.10.03-11.06.56.log](https://github.com/microsoft/AirSim/files/3684293/TaiYangShenMiao-backup-2019.10.03-11.06.56.log)

The above file is the log file. And I used administrator mode

rajat2004, 2019-10-04:
@sbkybhy Is the default Blocks environment working properly? If not, then that's the first problem which needs to be fixed before moving to custom environments.

You can try some more things such as running `clean_rebuild.bat` and then starting over,
The problems from the log file seem to be this - 
```
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
```
Also -
```
LogInit: Warning: Incompatible or missing module: UE4Editor-TaiYangShenMiao.dll
LogInit: Warning: Incompatible or missing module: UE4Editor-AirSim.dll
Running D:/EpicGames/ue18/UE_4.18/Engine/Binaries/DotNET/UnrealBuildTool.exe taiyangshenmiao Development Win64 -project="D:/Epic/taiyangshenmiao/taiyangshenmiao.uproject" -editorrecompile -progress -NoHotReloadFromIDE
Creating makefile for taiyangshenmiao (no existing makefile)
ERROR: Expecting to find a type to be declared in a target rules named 'taiyangshenmiaoTarget'.  This type must derive from the 'TargetRules' type defined by Unreal Build Tool.
```

Can you share the edited `.uproject` file? The error about the missing `.dll` files seems to be a widespread one in Unreal, with variety of possibilities, removing the `Intermediate` and `Binaries` folder and trying again might work

Hope this helps!

sbkybhy, 2019-10-06:
I solved my problem. After running "clean_rebuild. Bat", I added "Windows SDK8.1" to vs2017 by reading log file. Thank you for your help！

rajat2004, 2019-10-06:
@sbkybhy Great!

closed 2019-10-07 by ironclownfish

kaushikankur, 2020-03-24:
Install windows sdk 8.1 , then run clean.bat and again run update_from_git.bat