microsoft/AirSim #2220: move on path

oooohhhhhhh, 2019-10-01:
Hi , i'm trying to move the multirotor on path .
I run the code on https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/path.py , it seems like to give it coordinate to achieve moving on path , but after taking off , my multirotor only go straight , it can't move to next coordinate and it can't stop.

appreciate

YumaTheCompanion, 2019-10-01:
Did you modify the original code?

Seems like you increased the amount of wait, so it never passes to execute the other points.

Maybe your python program detaches from the simulation.

oooohhhhhhh, 2019-10-01:
I modified the original code, but it didn't work well

closed 2019-10-03 by oooohhhhhhh

nightduck, 2021-06-02:
@oooohhhhhhh did you find the line of code you editted? I'm getting these symptoms too: call takeoff and the drone ascends indefinitely. But I get these symptoms in both python and C++ programs. And I'm confused as to where my modification would be.

This is in a fork of AirSim and this behaviour is confirmed to be not present in vanilla AirSim on my machine. But all the relevant parts of AirLib are seemingly untouched. Video of behaviour attached when calling takeoffAsync and moveToPositionAsync

https://user-images.githubusercontent.com/10158631/120447764-891b4d80-c350-11eb-9a66-de7dcd7bb57b.mp4

https://user-images.githubusercontent.com/10158631/120447822-959fa600-c350-11eb-8f8b-1a510a2d2fbf.mp4

quikssb, 2023-05-14:
> @oooohhhhhhh did you find the line of code you editted? I'm getting these symptoms too: call takeoff and the drone ascends indefinitely. But I get these symptoms in both python and C++ programs. And I'm confused as to where my modification would be.
> 
> This is in a fork of AirSim and this behaviour is confirmed to be not present in vanilla AirSim on my machine. But all the relevant parts of AirLib are seemingly untouched. Video of behaviour attached when calling takeoffAsync and moveToPositionAsync
> 
>  infinite_takeoff.mp4 
>  wonky_flight.mp4

Did you find the solution? you would save my day

quikssb, 2023-05-14:
> @oooohhhhhhh did you find the line of code you editted? I'm getting these symptoms too: call takeoff and the drone ascends indefinitely. But I get these symptoms in both python and C++ programs. And [...]

Did you find the solution? you would save my day @oooohhhhhhh @nightduck