microsoft/AirSim #2223: Engine/Source/Runtime/XmlParser/ files not compiling

sohils, 2019-10-01:
### Issue
I have code that is referencing the XMLParser files to parse an XML document to create a robot from URDF files. This code sits in the Plugins/AirSim folder. On simply importing the project in Unreal, it is not able to compile my file die to a dependency on `XmlFile.h`. 

### Config
UE: 4.18
AirSim: 1.2

### Logs:
```bash
In file included from /home/simulation/Documents/UnrealEngine/Engine/Source/Runtime/XmlParser/Public/XmlFile.h:6:
/home/simulation/Documents/UnrealEngine/Engine/Source/Runtime/XmlParser/Public/XmlNode.h:12:1: error: expected expression
public:
^
/home/simulation/Documents/UnrealEngine/Engine/Source/Runtime/XmlParser/Public/XmlNode.h:10:21: error: variable has incomplete type 'class XMLPARSER_API'
class XMLPARSER_API FXmlAttribute
                    ^
/home/simulation/Documents/UnrealEngine/Engine/Source/Runtime/XmlParser/Public/XmlNode.h:10:7: note: forward declaration of 'XMLPARSER_API'
class XMLPARSER_API FXmlAttribute
      ^
/home/simulation/Documents/UnrealEngine/Engine/Source/Runtime/XmlParser/Public/XmlNode.h:36:2: error: expected expression
        friend class FXmlFile;
        ^
/home/simulation/Documents/UnrealEngine/Engine/Source/Runtime/XmlParser/Public/XmlNode.h:34:21: error: variable has incomplete type 'class XMLPARSER_API'
class XMLPARSER_API FXmlNode
                    ^
/home/simulation/Documents/UnrealEngine/Engine/Source/Runtime/XmlParser/Public/XmlNode.h:10:7: note: forward declaration of 'XMLPARSER_API'
class XMLPARSER_API FXmlAttribute
      ^

```
### What I have tried.
I tried enforcing clang version to 5.0.0. and the same error came up.

rajat2004, 2019-10-04:
@sohils It's not exactly clear if the problem is with AirSim itself and is difficult to find out without knowing the changes, but will leave that upto the maintainers who will be able to help you in a better way

Also, have you looked at this excellent fork of AirSim for URDF models - https://github.com/mitchellspryn/UrdfSim
Might be what you're looking for

sohils, 2019-10-04:
Yes @rajat2004. I am trying to run that fork. But the UrdfSim version is fork of an older version of AirSim. I spent time integrating the UrdfSim plugins with the latest version of AirSim and am now facing this issue where `XmlNode.h` is being imported.

rajat2004, 2019-10-05:
@sohils Ah, so you're trying to bring the UrdfSim repo upto date with the latest master
Maybe opening an issue on the Urdf repo might be useful

The error is strange, in that something seems wrong in the Unreal Engine code itself at first glance. Also, it's not clear where exactly is the `XmlFile.h` being included in AirSim, though didn't look too much into it..

 Hopefully, the maintainers will be more useful here, though will post if anything comes up, all the best with this!

ajberlier, 2020-01-26:
I am having the same issues. Any chance you found a solution?

mitchellspryn, 2020-01-27:
Two things:

1) UrdfSim does not build on linux. This is a [known issue](https://github.com/mitchellspryn/UrdfSim/issues/8). Unfortunately, I do not have the time or a linux build machine to fix this issue, so there is no ETA for a fix.

2)  If you do manage to get a windows machine, UrdfSim now uses engine version 4.24. I don't recall it being broken on 4.18, but I haven't explicitly tested it.

Having said that, try taking a look at [UrdfParser.h and UrdfParser.cpp here](https://github.com/mitchellspryn/UrdfSim/tree/master/Unreal/Plugins/AirSim/Source/Vehicles/UrdfBot/UrdfParser), this is what UrdfSim uses. Perhaps there's some syntax error in your code? It would be easier to debug if you posted the file you were trying to use.

ajberlier, 2020-01-27:
I am now working on Windows. I have run the newest version with UrdfSim with 4.18, so it does work well. 

When I tried merging with the newest version of AirSim to integrate with a my teams fork of AirSim that has diverged from AirSim, I am getting this same error on Windows.

I am getting the error for every instance of FXmlNode in both UrdfParser.h and UrdfParser.cpp

"error C2061: syntax error : identifier 'FXmlNode' " for line 27 in UrdfParser.h
UrdfLinkSpecification* ParseLinkSpecification(**FXmlNode** *linkNode);

mitchellspryn, 2020-01-27:
I just remembered one more detail.

I think you have to add "XmlParser" as a public dependency in the build.cs. Check [the UrdfSim implementation](https://github.com/mitchellspryn/UrdfSim/blob/master/Unreal/Plugins/AirSim/Source/AirSim.Build.cs). Look at line 80, the PublicDependencyModuleNames.AddRange() call.