microsoft/AirSim #2226: memory error

oooohhhhhhh, 2019-10-03:
I tried to run DQNdrone.py
(https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/DQNdrone.py), but it got error(MemoryError: Unable to allocate array with shape (500000, 84, 84) and data type float32).
I wonder if you can tell me what mean it is.

appreciate

rajat2004, 2019-10-04:
A quick calculation shows it's trying to allocate roughly 13 GB of memory, possibly RAM size is smaller.
```
>>> (1.0 * 500000 * 84 * 84 * 32) / (8 * 1024 * 1024 * 1024)
13.142824172973633
```

Try modifying the values here  - https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/DQNdrone.py#L254

I haven't actually ran this till now, so just a suggestion, hope it helps!

oooohhhhhhh, 2019-10-05:
Thank you i will try it later

closed 2019-10-05 by oooohhhhhhh