microsoft/AirSim #2229: How to make object detection dataset

vxgu86, 2019-10-04:
Object detection dataset need a bounding box around the object instead of segmentation mask.
Any hint to get this bounding box straight out?
Thanks

msb336, 2019-10-04:
A quick google search for "image mask to bounding box" returned [this tutorial](https://www.kaggle.com/voglinio/from-masks-to-bounding-boxes)

The basic idea is that if you know the pixel area of your classifier max, it is relatively easy to convert into a box by finding the two pixels the represent the bounding corners of the bounding box:
p(min_x, min_y), p(max_x,max_y)

closed 2019-10-04 by msb336

NickPerezCarletonUniversity, 2021-05-05:
> Object detection dataset need a bounding box around the object instead of segmentation mask.
> Any hint to get this bounding box straight out?
> Thanks

@vxgu86 did you get this working? maybe you can create a brief tutorial?