microsoft/AirSim #2230: The weather API cannot be used

sbkybhy, 2019-10-06:
The weather API cannot be used. I use "CityEnviron" engineering，which was downloaded from "https://github.com/microsoft/AirSim/releases/tag/v1.2.0". After unzipping, the document is ready to use. In this project, I can use python to control the car,but I cann't use python API to control weather.
 When I open the project, like the  below picture ( I  have put it into "computer_vision" mode ),
[image]
if I use "python weather.py", it reports errors like this "Traceback (most recent call last):
  File "weather.py", line 8, in <module>
    client.simEnableWeather(True)
  File "D:\AirSim\AirSim-master\PythonClient\airsim\client.py", line 82, in simEnableWeather
    return self.client.call('simEnableWeather', enable)
  File "C:\program1\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\program1\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'simEnableWeather' with argument count 1."
[image]

rajat2004, 2019-10-07:
Weather API was added after this version was released (https://github.com/microsoft/AirSim/commit/b1260c2c3177c886549a962094f286d6c45aac93)
You'll have to use a later release or master

sbkybhy, 2019-10-08:
Thank you, I solved the problem.  (*^▽^*)

rajat2004, 2019-10-08:
@sbkybhy Great! Could you close the issue if the problem has been solved?

closed 2019-10-08 by sbkybhy