microsoft/AirSim #2232: Can not compile AirSim at Linux (lack of clang++)

yingshaoxo, 2019-10-08:
The OS I'm running is: 
```
Linux pop-os 5.0.0-21-generic #22+system76-Ubuntu SMP Tue Jul 16 19:57:52 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```
___

The build command is:
```
./setup.sh
./build.sh
```

The error I'm getting is:
```
************************************
AirSim setup completed successfully!
************************************
+ ./build.sh
+ MIN_GCC_VERSION=6.0.0
+ gccBuild=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ uname
+ '[' Linux == Darwin ']'
++ readlink -f cmake_build/bin/cmake
+ CMAKE=/home/yingshaoxo/AirSim/cmake_build/bin/cmake
+ false
+ [[ ! -d llvm-source-50 ]]
+ [[ -d ./llvm-build/output/lib ]]
+ echo 'ERROR: clang++ and libc++ is necessary to compile AirSim and run it in Unreal engine'
ERROR: clang++ and libc++ is necessary to compile AirSim and run it in Unreal engine
+ echo 'Please run setup.sh first.'
Please run setup.sh first.
+ exit 1
```
___

But I just installed `clang` and `libc++` by:
```
sudo aptitude install clang 
sudo apt-get install libc++-dev
```
___

The version of clang installed at my computer is:
```
yingshaoxo@pop-os:~/AirSim$ clang -v
clang version 8.0.0-3 (tags/RELEASE_800/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/9
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/8
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/9
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/9
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7.4.0
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/8
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/9
Selected GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/8
Candidate multilib: .;@m64
Selected multilib: .;@m64
Found CUDA installation: /usr/lib/cuda-10.0, version 10.0
```
___

The output for `sudo apt-get install -y clang-5.0 clang++-5.0` is:
```
The following packages have unmet dependencies:
 clang-5.0 : Depends: libstdc++-5-dev but it is not installable
             Depends: libgcc-5-dev but it is not installable
             Depends: libobjc-5-dev but it is not installable
 libclang-5.0-dev : Depends: libstdc++-5-dev but it is not installable
                    Depends: libgcc-5-dev but it is not installable
                    Depends: libobjc-5-dev but it is not installable
E: Unable to correct problems, you have held broken packages.
```

___

The output for `sudo aptitude install -y clang-5.0 clang++-5.0` is:
```
Couldn't find any package whose name or description matched "clang++-5.0"
Unable to apply some actions, aborting
```

madratman, 2019-10-08:
That's weird.
Setup.sh should install clang for you.
One workaround is to set cxx compiler to clang++ manually
`export CXX=/usr/bin/clang++-5.0`
`./build.sh`

madratman, 2019-10-08:
It seems like setup.sh failed for you. Run it again, and you should see an error somewhere in the terminal output. Please post it here when you do.

yingshaoxo, 2019-10-10:
I have tried the same thing at the official Ubuntu. It works well.

So I think the main problem for this is that not every Linux distributions have clang-5.0 available for installing.

Maybe clang-5.0 is out-of-date for now. I would like to suggest the AirSim team to update their c library to clang-6.0 or higher.

Remember, in the software world, the newer, the better.

farhanhubble, 2019-10-19:
@yingshaoxo I just ran into this problem when trying to install AirSim on Fedora 29., although I had clang 5.0 installed in `/usr/local/bin`. So I edited two lines in bothe setup.sh and build.sh which set the compiler for linux to this:

```
        export C_COMPILER=$(which clang)
        export COMPILER=$(which clang++)

```

yingshaoxo, 2019-10-19:
@farhanhubble 

It's not about that, I just wonder if we can build it with clang++6.0 or higher. Because newer Linux don't have clang++5.0 available for installing.

shuyaosong, 2019-10-30:
@yingshaoxo   @farhanhubble 
I want to know whether you have solve such problem with this method.I also meet same error,thanks

jackfong123, 2019-12-03:
mkdir -p llvm-build/output
wget http://releases.llvm.org/4.0.1/clang+llvm-4.0.1-x86_64-linux-gnu-debian8.tar.xz
tar -xf "clang+llvm-4.0.1-x86_64-linux-gnu-debian8.tar.xz" -C llvm-build/output

solved the problem.

jackfong123, 2019-12-03:
```
mkdir -p llvm-build/output
wget http://releases.llvm.org/4.0.1/clang+llvm-4.0.1-x86_64-linux-gnu-debian8.tar.xz
tar -xf "clang+llvm-4.0.1-x86_64-linux-gnu-debian8.tar.xz" -C llvm-build/output
```

solved the problem.

madratman, 2020-04-03:
current master uses clang-8
either way, setup.sh and build.sh did work before. seems to me you somehow deleted `llvm-source-50` post running setup.sh

closed 2020-04-03 by madratman