microsoft/AirSim #2240: PX4 VTOL model support

sunnyroad, 2019-10-10:
Hi, 

Does AirSim currently support  PX4 VTOL model ?

If not, would you help add and support PX4 VTOL model such as "HIL Standard VTOL QuadPlane" and "Generic Quadplane VTOL" ?

petergerten, 2019-10-19:
I am also very interested in VTOL support. Did you get further with this yet?

sunnyroad, 2019-10-21:
> I am also very interested in VTOL support. Did you get further with this yet?

No. Can any guys help support this?

petergerten, 2020-02-07:
I wonder if anyone at least tried this.
The original paper mentions 

> "By allowing the movements of these vertices during the flight it is possible to simulate the vehicles with capabilities such as Vertical Take-Off and Landing  (VTOL) and other recent quadrotors that change their configuration in flight."

petergerten, 2020-08-27:
Hopefully the work here https://github.com/microsoft/AirSim/issues/2955 can be helpful for VTOL as well.

jonyMarino, 2021-06-01:
See: https://github.com/AOS55/Fixedwing-Airsim
and continue in #2955 please

closed 2021-06-01 by jonyMarino