microsoft/AirSim #2245: Call to 'getCarControls()' terminate called after throwing an instance of 'rpc::rpc_error'

chriscreber, 2019-10-10:
The call to msr::airlib::getCarState() works perfectly fine, but the newly added msr::airlib::getCarControls() returns with an rpc:rpc_error. I want to record the controls for deep learning purposes, but as of now I'm at a stand still. Am I calling this incorrectly, or is there an issue with the function? All help is appreciated. 

Code:
int main(int argc, char *argv[]) {
    CarRpcLibClient client;
    client.confirmConnection();
    CarApiBase::CarState state = client.getCarState(<carName>);
    CarApiBase::CarControls controls = client.getCarControls(<carName);
}

Error:
terminate called after throwing an instance of 'rpc::rpc_error'
  what():  rpc::rpc_error during call
Aborted (core dumped)

I'm running on Ubuntu 16.04 with the most up to date Airsim version on Unreal 4.18. Also, I'm trying to stick with the C++ library.

chriscreber, 2019-10-13:
So to answer my own question... I couldn't just pull to update I had to reinstall unreal and airsim. I'm guessing I did a few too many steps, but it is working now.

saihv, 2020-01-20:
Closing as issue seems to be resolved.

closed 2020-01-20 by saihv