microsoft/AirSim #2246: Airsim with PX4

cnun02, 2019-10-11:
[image]
`{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "LogViewerHostIp": "192.168.0.4",
      "LogViewerPort": 14388,
      "OffboardCompID": 1,
      "OffboardSysID": 134,
      "QgcHostIp": "192.168.0.4",
      "QgcPort": 14550,
      "SerialBaudRate": 115200,
      "SerialPort": "*",
      "SimCompID": 42,
      "SimSysID": 142,
      "SitlIp": "192.168.0.4",
      "SitlPort": 14556,
      "UdpIp": "127.0.0.1",
      "UdpPort": 14560,
      "UseSerial": false,
      "VehicleCompID": 1,
      "VehicleSysID": 135,
      "Model": "Generic",
      "LocalHostIp": "192.168.0.4"
    }
  }
}`
I have this error...
my environment,.
PX4 v5
UE4 4.18 + airsim
and I used wireless Lan

rajat2004, 2019-10-11:
Your settings are for SITL, rather than for HITL
For HITL, you'll have to set `UseSerial` to false

cnun02, 2019-10-11:
> Your settings are for SITL, rather than for HITL
> For HITL, you'll have to set `UseSerial` to true

Thanks...I'll try it

cnun02, 2019-10-11:
> Your settings are for SITL, rather than for HITL
> For HITL, you'll have to set `UseSerial` to true

[image]

Now I have new error..

closed 2019-10-11 by cnun02

cnun02, 2019-10-14:
This problem was solve!
1. Change PX4 v5 -> PX4 v2
2. QgroundControl must off.
3. not change the 'settings.json'

Rufio-Boj, 2019-10-22:
I had the same problem as you guys and found a way of making it work with PX4 v5 using Windows without the need to change to v2.
In the settings.json file, I changed _"SerialBaudRate":11520_ to **"SerialBaudRate": 921600**  and "SerialPort="*" to the actual port that my computer is using (Look it up in the Device Administrator). In my case, for example, it was **"SerialPort"=COM4.**

Hope this helps!

cnun02, 2019-10-23:
Thank you, Friend!!!
it does work!!!!!!!

Rufio-Boj, 2019-10-23:
Glad to help! :)