microsoft/AirSim #2251: self.simGetCollisionInfo().object_name always return (null)

vxgu86, 2019-10-14:
if self.simGetCollisionInfo().has_collided == True: 
                print('!collision_info',self.simGetCollisionInfo().object_name)

always print (null),  I did not know how to debug this?

ironclownfish, 2020-08-14:
@vxgu86 can you tell us more about the circumstances in which this happened? Just in case it's a function of being in CV mode, colliding with a certain thing, etc

ironclownfish, 2020-08-14:
@vxgu86 Please feel free to open a new issue if you still have this problem, and include the following
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

Thanks!

closed 2020-08-14 by ironclownfish