microsoft/AirSim #2257: Error at startup: Failed to load asset class

ZhanLiao, 2019-10-16:
hello everyone,,when i press F5 to run LandscapeMountains.sln, this will start the Unreal Editor.then ,PlayerStart object. something have wrong in following: Error at startup: Failed to load asset class - Class'/AirSim/VehicleAdv/SUV/SuvCarPawn.SuvCarPawn_C'

rajat2004, 2019-10-17:
Is the default Blocks environment working correctly? If not, then that would be the first thing to fix.
Are you using Windows or Linux? 

Try rebuilding after using `clean.sh` or `clean.cmd`, if that doesn't work, then run `clean_rebuild.sh` which will clean out everything and redownload the assets, which I think should fix the problem
Hope this helps!

ZhanLiao, 2019-10-18:
default Blocks environment working wrongly，when run block.sln, this displays：can't find UE4.exe

ZhanLiao, 2019-10-18:
> Is the default Blocks environment working correctly? If not, then that would be the first thing to fix.
> Are you using Windows or Linux?
>
> Try rebuilding after using `clean.sh` or `clean.cmd`, if that [...]

closed 2019-10-18 by ZhanLiao

dongya-bit, 2020-12-04:
> 大家好，当我按F5键运行LandscapeMountains.sln时，这将启动Unreal Editor.then，PlayerStart对象。以下内容有误：启动时出现错误：无法加载资产类-类'/AirSim/VehicleAdv/SUV/SuvCarPawn.SuvCarPawn_C'

I have the same problem. Have you solved your problem

huolongguoc, 2025-07-11:
I've met the same problem, I tried to delete C:/Users/Document/AirSim/settings.json and started the project again then it worked.