microsoft/AirSim #2261: How can I inject my modified version of AirSim plugin to released environments?

seheevic, 2019-10-18:
I am implementing and testing 'lockstep feature' into AirSim.
(It seems that this already has been asked and requested here)
I hope this task will save much time during training with open AI gym environment without sacrifice of physics accuracy.
I figured out some possibility and of course further works is required for now, but I don't know how to inject my new AirSim code (or lib) to released AirSimNH or City environment.
What I can do and test is only 'Blocks' in which I have to do some level design from scratch.

And as a beginner of Unreal Engine, is it possible to create same named project (like AirSimNH), then package it, and then just switch the '.pak' file with released one (which seems to contain the level assets)?
Please shed more light on me :)

seheevic, 2019-10-18:
Woops! Switching '.pak' file is working!

closed 2019-10-18 by seheevic

fawadahm, 2019-10-22:
@seheevic , could you please give a step by step guide on how you were able to do this?

seheevic, 2019-10-28:
@fawadahm 
Firstly, you need to look around the official reference to create custom environment
https://microsoft.github.io/AirSim/docs/unreal_custenv/
Some of the steps described below is just copy-past (or modified a little) from this official guide.
But since packaging with large asset (like LandscapeMountains as described in the guide) would take long time, I started my environment from scratch.
Note my Unreal Engine version is 4.22 which supports VS2019 and my OS is Windows 10. 

1. Create Blueprint Unreal Project named 'AirSimNH' in '.../AirSim/Unreal/Environments' from UE4Editor, so that the full path of 'AirSimNH.uproject' file is '.../AirSim/Unreal/Environments/AirSimNH/AirSimNH.uproject'.  I chose 'Flying' template 'With Starter Content'.
2. Open AirSimNH.uproject, it should launch the Unreal Editor
3. From the File menu select New C++ class, leave default None on the type of class, click Next, change default name as 'AirSimNH', and click Create Class. We need to do this because Unreal requires at least one source file in project. It should trigger compile and open up Visual Studio solution AirSimNH.sln.
4. Copy the code of .../AirSim/Unreal/Environments/Blocks/Source/Blocks/Blocks.[cpp|h|Build.cs] (from git repo) to newly created AirSimNH.[cpp|h|Build.cs] (3 files). Note we have to change the word 'Blocks' in the copied files to 'AirSimNH' which is our target project name.
5. Copy update_from_git.bat from ".../AirSim/Unreal/Environments/Blocks" to ".../AirSim/Unreal/Environments/AirSimNH" and run it.  This way now your own Unreal project has AirSim plugin.
6. Edit the AirSimNH.uproject to the same as ".../AirSim/Unreal/Environments/Blocks.uproject" (but of course, leave the word 'AirSimNH' as it be in uproject file)
7. Close Visual Studio and the Unreal Editor and right click the AirSimNH.uproject in Windows Explorer and select Generate Visual Studio Project Files. This step detects all plugins and source files in your Unreal project and generates .sln file for Visual Studio.
8. Reopen AirSimNH.sln in Visual Studio, and make sure "DebugGame Editor" and "Win64" build configuration is the active build configuration.
9. Press F5 to run. This will start the Unreal Editor. Remove auto generated 'NetworkPlayerStart' in World Outliner panel. Then create new PlayerStart object. (you can do this by finding PlayerStart class from 'Place' tab and just drag that class to the position you want on the viewport) Make sure its location is valid vehicle position. This is where AirSim plugin will create and place the vehicle. If its too high up then vehicle will fall down as soon as you press play giving potentially random behavior. (but actually, we will replace this to released data, so this validitiy is only effective to test purpose)
10. In Window/World Settings, set the GameMode Override to AirSimGameMode (you can see the concrete image guide in the official reference above)
11. Go to 'Edit->Editor Preferences' in Unreal Editor, in the 'Search' box type 'CPU' and ensure that the 'Use Less CPU when in Background' is unchecked. If you don't do this then UE will be slowed down dramatically when UE window loses focus.
12. Be sure to Save these edits. Hit the Play button in the Unreal Editor.
13. Skippable - Test in Visual Studio (open sln file, run on DebugGame Editor/Win64)
14. Open Unreal Editor, Execute [File/Package Project/Windows/Win64] from top menu.
15. Replace .pak file in the generated package ".../WindowsNoEditor/AirSimNH/Content/Paks/AirSimNH-WindowsNoEditor.pak" to the .pak file from released zip.
16. Done! 14,15 can be automated by batch command file.

I am organizing this from my memory, so something might be wrong ;(
If you find out something wrong, please let me know.
I hope this could help you.

PS, If this has some information against this AirSim repo and liscense, let me know. :(

fawadahm, 2019-10-31:
@seheevic , thank you for the detailed response :)