microsoft/AirSim #2263: Error: You Need to Install NullSourceCodeAccessor

farhanhubble, 2019-10-19:
I am building AirSim on Fedora 28. I had to compile clang-5.0 from source and then used it to compile Unreal, following the steps here: https://microsoft.github.io/AirSim/docs/build_linux/. I then built AirSim. When I launch Unreal Editor and browse the Blocks environment to open it, I get an error: 
```
To open this project you must first install NullSourceCodeAccessor. Would you like to install it now?
```
[image]

rajat2004, 2019-10-19:
This might help you - https://github.com/fire/SensibleEditorSourceCodeAccess
Source - https://answers.unrealengine.com/questions/262589/it-only-shows-vs-for-a-source-code-editor.html 

Also, it's mentioned on the next line in the section for Vim in the Unreal Engine Wiki-

If you are running an engine version that does not have this code accessor built in, run:
```
cd Engine/Plugins/Developer && git clone https://github.com/fire/SensibleEditorSourceCodeAccess
```

farhanhubble, 2019-10-19:
Thanks Rajat! However, after cloning the plugin, I get another error, right at load time, that says that the plugin could not be loaded because the relevant module was not found.

[image]

I will close this issue though because this is definitely an Unreal issue.

rajat2004, 2019-10-19:
Did you build Unreal Engine again after cloning the plugin? It's mentioned in the GitHub repo readme

farhanhubble, 2019-10-19:
Oops! I hadn't. After I rebuilt UE4Editor, the missing plugin error is also gone. Thanks a ton @rajat2004 . I am now getting a different error `/usr/bin/ld: cannot find -lsupc++` for which I'll open a ticket if I cannot  fix it myself.

closed 2019-10-19 by farhanhubble

aslepov01, 2020-02-07:
I have a similiar problem. After clonning https://github.com/fire/SensibleEditorSourceCodeAccess, patching Engine/Config/Linux/LinuxEngine.ini and rebuilding UE4Editor, I still have message "You Need to Install NullSourceCodeAccessor" while trying to open Blocks project. What  am I doing wrong?

fangchaooo, 2022-07-29:
because The 'Developer' module type has been deprecated in 4.24