microsoft/AirSim #2265: Error: cannot find -lsupc++ when linking libUE4Editor-AirSim.so

farhanhubble, 2019-10-19:
I am Using Fedora 28. I have build AirSim and Unreal engine with clang 5.0. Now I am trying to open the Blocks environment with Unreal Editor. I see a popup telling me to convert the project and when I select to convert in place, it kickstarts compiling some modules. The compilation results in a linking error:

[image]

I found a similar error [here](https://github.com/microsoft/AirSim/issues/678) and checked that the patch mentioned is already present in my code. 

I am not sure which Makefile is pulling in this library.
Here's what grep tells me:
```
farhanhubble@farhan-legion:~/farhan-devel/microsoft-airsim/AirSim$ grep -rn 'supc++'
Unreal/Plugins/AirSim/Source/AirSim.Build.cs:129:			PublicAdditionalLibraries.Add("supc++");
Unreal/Environments/Blocks/Plugins/AirSim/Source/AirSim.Build.cs:129:			PublicAdditionalLibraries.Add("supc++");
Binary file Unreal/Environments/Blocks/Intermediate/Build/Linux/BlocksEditor/Development/Makefile.ubt matches
Unreal/Environments/Blocks 4.18/Plugins/AirSim/Source/AirSim.Build.cs:129:			PublicAdditionalLibraries.Add("supc++");
Binary file Unreal/Environments/Blocks 4.18/Intermediate/Build/Linux/BlocksEditor/Development/Makefile.ubt matches
Binary file llvm-source-50/test/tools/sancov/Inputs/test-linux_android_aarch64 matches
llvm-source-50/projects/libcxx/CMakeLists.txt:111:set(CXXABIS none default libcxxabi libcxxrt libstdc++ libsupc++ vcruntime)
llvm-source-50/projects/libcxx/src/support/runtime/exception_glibcxx.ipp:12:#error header can only be used when targeting libstdc++ or libsupc++
llvm-source-50/projects/libcxx/src/support/runtime/exception_pointer_glibcxx.ipp:11:// libsupc++ does not implement the dependent EH ABI and the functionality
llvm-source-50/projects/libcxx/TODO.TXT:6:* Look into mirroring libsupc++'s typeinfo vtable layout when libsupc++/libstdc++
llvm-source-50/projects/libcxx/utils/libcxx/test/config.py:810:        elif cxx_abi == 'libsupc++':
llvm-source-50/projects/libcxx/utils/libcxx/test/config.py:811:            self.cxx.link_flags += ['-lsupc++']
llvm-source-50/projects/libcxx/cmake/Modules/HandleLibCXXABI.cmake:77:    "${LIBCXX_CXX_ABI_LIBNAME}" STREQUAL "libsupc++")
llvm-source-50/projects/libcxx/cmake/Modules/HandleLibCXXABI.cmake:87:    set(_LIBSUPCXX_LIBNAME supc++)
llvm-source-50/projects/libcxx/cmake/Modules/HandleLibCXXABI.cmake:124:     Currently libstdc++, libsupc++, libcxxabi, libcxxrt, default and none are
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:72:On Linux, it is also possible to use :ref:`libsupc++ <libsupcxx>` or libcxxrt.
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:258:  **Values**: ``none``, ``libcxxabi``, ``libcxxrt``, ``libstdc++``, ``libsupc++``.
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:379:Using libsupc++ on Linux
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:382:You will need libstdc++ in order to provide libsupc++.
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:384:Figure out where the libsupc++ headers are on your system. On Ubuntu this
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:420:You can also substitute ``-DLIBCXX_CXX_ABI=libsupc++``
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:421:above, which will cause the library to be linked to libsupc++ instead
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:424:GCC ships libsupc++ separately but only as a static library.  If a
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:426:own copy of libsupc++ and this can lead to subtle problems.
llvm-source-50/projects/libcxx/docs/BuildingLibcxx.rst:459:clang is set up to link for libc++ linked to libsupc++.  To get around this
llvm-source-50/projects/libcxxabi/lib/buildit:62:    # FIXME: removing libgcc and libsupc++ dependencies means porting libcxxrt and LLVM/compiler-rt
llvm-source-50/projects/libcxxabi/lib/buildit:66:        -lsupc++ -lpthread -lmingw32 -lgcc_s -lgcc -lmoldname -lmingwex -lmsvcr100 -ladvapi32 -lshell32 -luser32 -lkernel32 -lmingw32 -lgcc_s -lgcc -lmoldname -lmingwex -lmsvcrt"
cmake/cmake-modules/CommonSetup.cmake:78:                # removed -lsupc++ from below (Git issue # 678)
llvm-build/CMakeCache.txt:2059:LIBCXX_CXX_ABI-STRINGS:INTERNAL=none;default;libcxxabi;libcxxrt;libstdc++;libsupc++;vcruntime

```

rajat2004, 2019-10-19:
Maybe it's something specific to Fedora?
Another old AirSim issue which came up - https://github.com/Microsoft/AirSim/issues/601
Looking for Fedora specific things (Static library or something?) - https://github.com/numenta/nupic/issues/1901#issuecomment-97048452

Hope this helps!

farhanhubble, 2019-10-19:
I have been able to get around this problem by commenting out the lines:

```
Unreal/Plugins/AirSim/Source/AirSim.Build.cs:129:  PublicAdditionalLibraries.Add("supc++");
Unreal/Environments/Blocks/Plugins/AirSim/Source/AirSim.Build.cs:129:	 PublicAdditionalLibraries.Add("supc++");
```

rajat2004, 2019-10-20:
@farhanhubble Great! So it's working correctly now?
You could open a PR for fixing this if you want to, the devs could have a look at it
Also, if it's fixed, then maybe the issue can be closed

farhanhubble, 2019-10-29:
I am not familiar with the *.Build.cs scripts used by Unreal but for a good PR this line should be disabled selectively, only if the Clang compiler is being used.

madratman, 2020-04-13:
@farhanhubble a PR is welcome, but we don't support fedora

closed 2020-04-13 by madratman