microsoft/AirSim #2269: Poor flight performance when using Lidar in SITL mode

marufino, 2019-10-21:
There seems to be some performance issues when using the Lidar in SITL mode. When simulating a dense lidar cloud (1024*16) the flight performance degrades dramatically, the drone can only hover for a few seconds before spiraling out of control. This does not happen without the Lidar, and it also doesn't seem to be an issue when using SimpleFlight. The entire time the simulation is running in the 25-30 fps range.

Additionally, when subscribing to the lidar data, the size of the pointclouds is variable, and not the expected 16384 points, it hovers around 8000 instead(anywhere from 7000-9000 while the drone is static)

My guess is the simulation can't keep up with creating so many lidar points so quickly and that's causing it to run behind?

Any help would be appreciated

I'm running the latest Airsim master commit, Unreal Engine 4.18, on a Linux computer with the following specs:
CPU: i7-6850K CPU @ 3.60GHz
Memory: 126Gb
Video Card:Titan X, Driver Version: 418.40.04

settings.json
```
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "Sensors": {
        "Barometer": {
          "SensorType": 1,
          "Enabled": true
        },
        "Imu": {
          "SensorType": 2,
          "Enabled": true
        },
        "Gps": {
          "SensorType": 3,
          "Enabled": true
        },
        "Magnetometer": {
          "SensorType": 4,
          "Enabled": true
        },
        "Distance": {
          "SensorType": 5,
          "Enabled": true
        },
        "Lidar": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "RotationsPerSecond": 10,
          "PointsPerSecond": 163840,
          "X": 0,
          "Y": 0,
          "Z": 0,
          "Roll": 0,
          "Pitch": 0,
          "Yaw": 0,
          "VerticalFOVUpper": 16,
          "VerticalFOVLower": -16,
          "DrawDebugPoints": false,
          "DataFrame": "VehicleInertialFrame"
        }
      }
    }
  }
}
```

RaiMunoz, 2019-12-10:
Yea I wanted to ask about this as well. Is it because of the latency of the airsim rpc client or has this been looked into? Thanks!

saihv, 2020-01-20:
> This does not happen without the Lidar, and it also doesn't seem to be an issue when using SimpleFlight

Does it happen if Lidar is used along with Simple Flight?

marufino, 2020-01-27:
> > This does not happen without the Lidar, and it also doesn't seem to be an issue when using SimpleFlight
> 
> Does it happen if Lidar is used along with Simple Flight?

Simple Flight seems to work fine with lidar, in my testing, those same lidar settings only cause issues if the SITL is being used

marufino, 2020-01-31:
After the lockstep update in https://github.com/microsoft/AirSim/pull/2310 and upgrading to PX4 release/1.9 this seems to be fixed!

closed 2020-01-31 by marufino