microsoft/AirSim #2270: Using Qgroundcontrol with PX4 HITL

akhilahmed, 2019-10-21:
I have a question regarding the features supported for PX4 HITL. 

From the PX4 HITL setup documentation and demo video, it appears the drone can only be controlled with the RC transmitter. This would be the ideal setup for use cases involving manual flight mode. For the use cases involving mission mode, it would be easier to have the support for mission planner applications like QGroundControl. Would it be possible to work with QGC in a PX4 HITL simulation? 

I was able to get QGC to connect to the UDP port 14540 and run missions in the SITL simulation. But in a HITL simulation, QGC is not able to connect to the Pixhawk. 

So, here are my questions.
1. Is QGC currently supported in HITL simulation?
2. If the answer to 1. is yes, which UDP port do I have to configure QGC to connect to?
3. If the answer to 1. is no, how do I go about enabling the support for QGC? I believe there should not be any architectural limitations in AirSim to support QGC+HITL as this is supported in Gazebo.

Details of my setup:
Unity 2019 on Ubuntu 16.04
AirSim v1.2.0
Pixhawk running PX4 v1.9.2

Thanks!

jessearmand, 2019-10-30:
I have only used Unreal engine. QGC could be run together with HITL simulation, but it needs to be launched only **after** the vehicle is connected via PX4 (the Pixhawk board) to the Simulator.

My workflow at the moment is to quit QGC when I want to launch the simulator. Then, I relaunch QGC again if needed.

akhilahmed, 2019-10-30:
Thank you for your inputs @jessearmand 
Are you running AirSim on Ubuntu or on Windows?

jessearmand, 2019-10-31:
On Windows. Perhaps, you should try Unreal Engine instead.

lovettchris, 2019-11-06:
The reason QGC must be launched after the simulator is because QGC will, by default, steal the serial port connection to the Pixhawk.  I think you can stop QGC from doing this if you uncheck the "Autoconnect" checkbox for Pixhawk on the [QGC settings page](https://docs.qgroundcontrol.com/en/SettingsView/General.html).

jessearmand, 2019-11-07:
I see. That’s why there was a page in documentation to uncheck that box. I forgot where was it, and for what reason.

DemircanOnur, 2019-11-07:
Which command do you start for SITL ? For v1.8.2 and older version `make posix_sitl_ekf2 none_iris` is working but in v1.9.2, it doesn't work.

lovettchris, 2019-11-07:
Correct.  I'm working on v1.9.2 as we speak.  v1.9.2 added a new feature called "lockstep" that needs to be implemented in AirSim.

akhilahmed, 2019-11-10:
Thank you for your response Chris. I think connecting QGC in HITL is only possible with the Unreal Engine as Jesse has confirmed its functionality. Is there a plan to add this support for Unity Engine too?

jessearmand, 2019-11-10:
Well I could try with Unity Engine as well. But, it takes another time to setup everything. Have you did some testing on your side? @akhilahmed

akhilahmed, 2019-11-10:
I went through the code and realized where I was going wrong. I used to start the simulation first and then run QGC and attempt to connect QGC with the simulation, whereas the code expects QGC to be up and running before starting the simulation. And as mentioned in this thread earlier, QGC should be configured to only connect via UDP (by unchecking all other options in Autoconnect). In summary, QGC does work with both engines. 

Thank you all for your help.

saihv, 2020-01-20:
Closing as issue seems to have been resolved.

closed 2020-01-20 by saihv