microsoft/AirSim #2272: New build releases for Linux

fawadahm, 2019-10-22:
The build releases for Linux are for AirSim version 1.2.0. That version of AirSim has some bugs in its IMU code. The current AirSim code has fixed that bug but there are no Linux binaries for it. My question is; are there any plans to release Linux binaries for the latest AirSim version? The reason I am asking is that it is very tedious to build different environments in Windows and then import them to Linux.

Neabfi, 2020-01-30:
Any chance? The latest Linux release is one year old. That would be awesome to have a new one.

theold190, 2020-04-03:
A similar question was asked in #2477. And there were some answers and details on a new linux release.

@fawadahm , does that answer your question? Is there anything else you expect to be answered/provided in this issue?

closed 2020-08-10 by jonyMarino