microsoft/AirSim #2273: Duration in moveByVelocityZAsync() does not work when running script

dimikout3, 2019-10-23:
I am using the python API from implementing drone swarm logic. I tested the moveByVelocityZAsync() using a python interpreter and the duration seemed to perform correctly. However, when I am using a python script each individual drone moves with the specified vx and vy even after the specified duration is passed. If I use the .join() method, each drone is moving exactly for the specified duration. But it is very insufficient waiting for each drone to reach its position sequentially, it would save me a lot of time If I could move the drones in parallel. I also tried threading (calling each drone's moveByVelocityZAsync() in individual threads) but I get a RuntimeERror "IOLoop is already running", which makes sense. Lastly, I tried using multiple clients (each for one drone), but again as I expected it did not work. 

Am I missing something ?

saihv, 2019-11-04:
Could you please try building the newest version and testing with that? We just pushed a fix for the bug where the drone keeps moving even after the specified duration.

dimikout3, 2019-11-07:
Thank you very much. I will check your fix asap!

saihv, 2020-01-20:
Closing due to inactivity.

closed 2020-01-20 by saihv